letscontrolit/ESPEasy #3860: ESPEasy_ESP32_mega-20211105 - Password

croftsquirrel, 2021-11-23:
Hi. I have noticed that after flashing the OTA update on a esp32 with ESP_Easy_mega_20211105_normal_ESP32_4M316k that I'm unable to login with Chrome and Edge.  Internet Explorer still allows me to login with password set.

I first tried re-flashing OTA update for the previous build I had installed and was able to login again.  Then I removed the password and able to browse ESPEasy within Chrome and Edge.  A soon as I set the admin password I am then unable to login again, except in Internet Explorer.

Hope this helps someone experiencing the same problem and that there is hopefully a fix for this issue.

TD-er, 2021-11-23:
This is very strange, as I do have nodes here myself where I have set a password to test and I am able to login using Chrome.
Maybe some browser extension is causing this? Or some setting, like not allowing to login over non-HTTPS connection?

kuema, 2021-11-29:
I'm facing the same issue here after updating one of my ESP32 to 20211105. I tried both Chrome and Firefox on my Xubuntu box, but can't login with either of them.

I connected the serial console and tried to set an empty password using the command `Password,`, did `save` and then `reboot`, but without success. Is this the correct way to disable the password auth?

Other than that, I could only `reset` the node to factory default and reconfigure it. As soon as I set a password, the login fails.
I even flashed a completely new unit with this version, same issue.

One note: the ESP8266 version of 20211105 doesn't seem to be affected.

TD-er, 2021-11-29:
Can you also check with some other browser or on another platform? (e.g. on a mobile)
I do have the impression it may be related to browser security settings as it isn't happening here.

kuema, 2021-11-30:
Of course, I can check that later when I'm back from work.

Curious thing is, it happens to both Chrome and Firefox with this new release only, while it worked flawlessly before with both browsers. The browser consoles didn't log anything interesting either.
I saw it uses digest auth (for a while now), maybe good-old IE does something different here on the client side. I don't have an IE at my disposal, so I can't check that in particular.

Where there any major changes regarding the authentication?

For the meantime, how can I disable authentication to work around this? The docs weren't clear on that and `Password,` on the console didn't work.

Thanks for your support! 🙂

croftsquirrel, 2021-11-30:
Hi everyone,  I've just tried on mobile (Samsung Galaxy S21, Android 11).  Both Chrome and Samsung Internet also not allowing to log in with a password.  So the only one that I've been able to use successfully is Internet Explorer Desktop version 21H1.

TD-er, 2021-11-30:
Hmm I will check again with the mentioned specific bin version.
I am not aware of changes in the authentication, but will also look into related code.

I have not used IE for years, and I am using Chrome here on my Windows PC.

croftsquirrel, 2021-11-30:
Yes quite strange.  Maybe it is only certain bin's that it's affecting.  None of my previous versions of ESPEasy on ESP8266's have ever done this.  Only the release and bin as mentioned in the subject on a esp32.

TD-er, 2021-11-30:
Can you check the used SDK version as mentioned on the sysinfo page?

croftsquirrel, 2021-11-30:
Hi TD-er, it's ESP32 SDK v3.3.4-432-g7a85334d8

TD-er, 2021-11-30:
OK, so it may be related to the used SDK.

TD-er, 2021-11-30:
> Is there anyway around this? Does it mean the chip is outdated? Will there be support for these?

Well the version I'm using on my test setup is a later version of the SDK, and I cannot reproduce it here.
So that  could mean it has already been fixed and I simply can make a new build to have it fixed. :)
But to be sure I have to either provide you with a new test build, or make sure I can reproduce it myself (or both ;) )

croftsquirrel, 2021-11-30:
Sure, let us know if we can help.

kuema, 2021-11-30:
Seems like you're on some track regarding the different SDK version.

Just for completeness, I wasn't successful to login with other browsers on different Android platforms as well. I have tried Firefox, Chrome and the Amazon Silk browser.

uzi18, 2021-12-01:
@TD-er we should have availble command like: "Password,'actual password',0" to disable password

DeNB3rt, 2021-12-07:
> 

I'm facing the same, how can I get this solved? Can't use the device without password. Thanks for the info! How can I force a later SDK? mine is also  `ESP32 SDK v3.3.5-1`
Thx!

TD-er, 2021-12-29:
As just confirmed by @tonhuisman this will for sure be fixed when we update to ESP32 IDF 4.4
See #3845

PR #3845 (2022-04-27): [ESP32] Update to IDF 4.4 Arduino-ESP32 2.0.1
Including lots of WiFi fixes to be able to get a stable WiFi connection on the new ESP32 SDK.

This makes the bin file size quite a bit smaller (300k less on "Custom" build)
See this comment about the effects of the new bootloader in IDF 4.4: https://github.com/esphome/esp-web-tools/issues/151#issuecomment-1004296924

Fixes: #3860
Fixes: #3948

ToDo:

- [ ] Test I2C plugins  (see: https://github.com/espressif/arduino-esp32/pull/5683 & https://github.com/espressif/arduino-esp32/issues/6373#issuecomment-1059196867  & https://github.com/espressif/ardu [... cut]
closed 2022-04-27 by TD-er