letscontrolit/ESPEasy #3861: Naming of IR Receive plugin P016 analogue to IR Transmit P035

Wookbert, 2021-11-23:
For newbies and those not familiar with names of individual electronic components (which includes myself) I find/found it difficult to locate the Infrared receiving plug-in. At https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins it is listed as _Communication - TSOP4838_. As the TSOP4838 is just one of many 38Khz TSOPs, the naming is a bit short-sighted.

**Instead I'd suggest rename it throughout documentation and source code to IR Receive, given that P035 is IR Transmit (IRRX and IRTX where used abbreviated).**

TD-er, 2021-11-23:
Yep that makes sense, especially since the builds with IR related plugins is in the IR builds.

Wookbert, 2021-11-23:
### Source file naming
Currently it’s `_P016_IR.ino` and `_P035_IRTX.ino`. To be consistent, the first one should be renamed to `_P016_IRRX.ino`

tonhuisman, 2021-11-24:
The naming of plugin files hasn't always been very consistent, so changing that a) doesn't help in identifying the plugin from the UI, b) will break any external references to the file, like documentation (yes, that _can_ be changed, but hmm) and c) then we could/should rename some other plugin files too, which could break even open pull requests. So I'd not even want to try starting an operation like that.

Renaming the plugin description is an easy to do task, and will probably only 'break' screenshots in the documentation and maybe some external guides/walk-through's, but indeed quite feasible, so that's IMHO the only change needed here.

TD-er, 2021-11-24:
Yep, the .ino files are still left as .ino files even though I really would like to move everything into .h/.cpp files.
This has several reasons and external references is one of them.
Not only references we control. but also blog posts etc. Would be a shame to loose those.
One could argue that the IR plugins are mainly maintained by 1 user (not being me for a change ;) ) but on the other hand the IR plugins are apparently quite unique to ESPEasy so would be a pity if potential new users end up in a 404 when following links in a blog.

There are lots of .ino files in the project that would be easier to find if their filenames would have been different.
e.g. the "Output" plugin. I know it is the Domoticz MQTT helper, but that's not really suggested by the name.

And while we're at it... the controller .ino files are also not really descriptively named.

PR #3862 (2021-12-12): [P016] Change plugin title Communication - IR Receive (TSOP4838)
Changes the plugin-title to include 'IR Receive', in line with P035: 'IR Transmit'.
Kept the TSOP4838 part to be recognizable for existing users.

Resolves #3861
closed 2021-12-12 by TD-er