letscontrolit/ESPEasy #3863: ESP32-C3 support

chemmex, 2021-11-28:
Any plans on adding ESP32-C3 to ESPEasy? Arduino support is maturing, more dev boards and modules are coming, especially AI-Thinker's pin-compatible modules. It seems C3 has good chances to become a real successor for ESP8266.

TD-er, 2021-11-28:
Yep, I got my ESP32-C3 modules this week in the mail.
So plans for sure :)

The ESP32-C3-12F is very useful as it is electrical and mechanical compatible with the ESP-12F.
So can be replaced on a lot of existing boards out there.

But I have not yet started adding support.
@Jason2866 has once told me it is easier to support the C3 compared to S2, so I hope he's right as the S2 did take quite a lot of work.

Jason2866, 2021-12-26:
@TD-er  Esp32 core 2.0.2 for Platformio (ESP32, C3, S2) to use only one entry!
`platform                    = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2/platform-tasmota-espressif32-2.0.2.zip`

TD-er, 2021-12-26:
Thanks :)
This also includes the changes for serial? (modbus specific changes)

TD-er, 2021-12-27:
@Jason2866 There is still the bug in it regarding the max size free block, which is just a few bytes short of 64k.

Jason2866, 2021-12-28:
@TD-er  Yes, since you are still using the old core ;-)
Remove the entry(s) for `platform_packages`  Everything is now in `platform` !

TD-er, 2021-12-28:
@Jason2866 
Thanks, that did indeed fix it.

I noticed the system libraries now show:

System Libraries: | ESP32 SDK v4.4-2.0.2
-- | --

Just to be sure, that "2.0.2" is now the espressif/arduino-esp32 tag, or do you keep track of another versioning nr?
I find the SDK versioning quite confusing and I guess if it is now showing IDF4.4 and `espressif/arduino-esp32` tag 2.0.2, then it is a huge improvement in numbering :)

TD-er, 2021-12-28:
@Jason2866 I am testing the platform you linked, but every time I try to run an ESP32 build with Little FS, the unit keeps crashing.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12380
ho 0 tail 12 room 4
load:0x40080400,len:2912
entry 0x400805c4
ets Jul 29 2019 12:21:46
```

Any idea what may be wrong here? (or changed in IDF 4.4)
It is a watchdog reset, so I will also tried with a smaller LittleFS file system.
Both with ~300k LittleFS and 8M (on 16M flash) give these reboots.
When using SPIFFS it runs just fine.
I also removed the defined tool-mklittlefs package from my platformIO config and also completely cleared the build env and related folders on the hard drive.

Jason2866, 2021-12-28:
@TD-er  The version numbering is my own. Not official. I am confused to how platformio does. And yes it is just Arduino Core version. I have no idea why LittleFS is not working for you. We use LittleFS as default filesystem and it just works.

TD-er, 2021-12-28:
Like I said, it was working fine, but I don't think it even tries to attempt to access the file system.
Just as if it tries to access something that isn't present in the binary.
Nothing showing in the logs and also not working on a node which does have a filesystem present.

TD-er, 2021-12-29:
I just made a test build for 4M builds with LittleFS and that one works just fine.
So it seems the `LittleFS.begin()` call may take too long to complete and thus cause a watchdog reset. (or the format... no logs :( )

TD-er, 2023-05-02:
See my comment here: https://github.com/letscontrolit/ESPEasy/issues/4614#issuecomment-1530615205

PR #4640 (2023-05-10): [ESP32-C3 / -S3] Add support for ESP32-C3 & ESP32-S3
#fixes: #4614
#fixes: #3863

ToDo ESP32-C3:
- [x] Implement pin definitions
- [x] Check settings struct for boot states
- [x] Implement hardware specifics (e.g. USB behavior, SPI busses)
- [x] Add proper defaults for SPI busses etc.
- [x] Fix determine free stack
- [x] Document
- [x] Test

ToDo ESP32-S3:
- [x] Implement pin definitions
- [x] Check settings struct for boot states
- [x] Implement hardware specifics (e.g. USB behavior, SPI busses)
- [x] Add proper defaults for SPI busses etc.
- [x] Document
- [x] Test

N.B. support for CDC (con [... cut]
closed 2023-05-10 by TD-er