letscontrolit/ESPEasy #3869: wESP32 RTL8201

devdelay, 2021-12-06:
Looking to add support for newer versions of [wESP32](https://wesp32.com/) that use ETH_TYPE as RTL8201 instead of LAN8720.

I've been able to configure the device through wifi using instructions on the forum [here](https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=8094#p48195) with ESP_Easy_mega_20211105_max_ESP32_16M1M_ETH-factory.bin

Current builds do not appear to have the option to change ETH_TYPE to RTL8201

TD-er, 2021-12-06:
When (as in date) has this design of the wESP32 boards changed?
I do have some of those boards here and those work just fine with ESPEasy.
I found [here](https://wesp32.com/hardware/) on the website this change was done at revision 7 apparently and to be honest I really don't like this change as I have made boards supporting this wESP32 board in a project. The whole idea was that it was better to pick a non-AliExpress board as there may be more guarantees for continuous product designs.

It probably has something to do with the global chip shortages, but still I don't like this :(

Well on a positive note for you, this means I have to make it work quite soon.

The site mentions improved reliability and I think that's not a bad thing as I already have 2 of these wESP boards which don't seem to have a working ethernet chip anymore.

devdelay, 2021-12-06:
@TD-er not sure when they changed this. I received two units last week via Amazon and both have RTL8201. From the updated documentation it seems they likely made the change due to 8720 being prone to damage with certain settings on their hardware while RTL8201 is not.

TD-er, 2021-12-06:
Well I have used the LAN8720 on my board designs too, but a few months ago I had to change it since the chip had a lead time of over 80 weeks.
It has essentially become part of the big unobtainium.

tonhuisman, 2022-05-07:
This seems to be solved now that IDF 4.4 has been merged into mega.

TD-er, 2022-05-07:
I have only added support for those, but I have never tested them myself.
I will close it, as it is indeed added, but if anyone ever tested them please let me know.

closed 2022-05-07 by TD-er