letscontrolit/ESPEasy #3870: Cannot connect to hidden AP on channels 12 / 13

ghtester, 2021-12-08:
Ì have just upgreaded 2 ESP nodes to Custom Vagrant build : ESP_Easy_mega_20211202_custom_IR_ESP8266_4M1M Dec 2 2021.
Unfortunately the both nodes couldn't connect to hidden AP anymore after the reboot (even after cold boot).
As soon as I enabled the SSID broadcast on the AP or after the AP channel was changed to 11, both nodes connected successfully to the same AP quickly.
Another ESP node (ESP_Easy_mega_20210801_custom_IR_ESP8266_4M1M Aug 1 2021) successfully reconnected to hidden AP on all tested channels (11,12,13).
On all three nodes the Fallback SSID is configured (the Fallback APs with weak signal or even missing).

TD-er, 2021-12-08:
Can you check the SDK version of both builds?
This is shown on the sysinfo page.

ghtester, 2021-12-08:
<html>
<body>

Firmware
--
Build:⋄ | 20116 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Dec 2 2021 12:16:54
Plugin Count:⋄ | 37 [IR]
Build Origin: | Vagrant
Build Time:⋄ | Dec 2 2021 12:15:09
Binary Filename:⋄ | ESP_Easy_mega_20211202_custom_IR_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega_549f76e

</body>
</html>

The older FW:
<html>
<body>

Firmware
--
Build:⋄ | 20114 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Aug 1 2021 10:36:17
Plugin Count:⋄ | 35
Build Origin: | Vagrant
Build Time:⋄ | Aug 1 2021 10:35:18
Binary Filename:⋄ | ESP_Easy_mega_20210801_custom_IR_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega_04cbbe3

</body>
</html>

TD-er, 2021-12-09:
OK, so the SDK is the same.
Only thing I can think of is that the scan I do myself differs from the scan made by the WiFi library.

What has changed (I guess it might even be based on your comments ;) ) is this.
Previous behavior:
- Scan all channels and collect all APs + sort them
- Filter out the ones matching + all hidden APs.
- Try to connect to all matching SSIDs using BSSID/channel as found during the scan
- Try to connect to all hidden SSIDs using all permutations of known keys using BSSID/channel as found during the scan

New behavior:
- Scan all channels and collect all APs + sort them
- Filter out the ones matching + all hidden APs.
- Try to connect to all matching SSIDs using BSSID/channel as found during the scan
- Just try all (max 2) SSID/key pairs to connect, without specifying BSSID and channel

So this last step is only using the Arduino call to `WiFi.begin()` using the SSID and key, not using the BSSID/channel.
This means it probably needs to do an attempt on all channels to see when an AP will accept this SSID/key combo.
Apparently this does not include channel 12 and 13 (and 14).
This could be a regional setting thing. But I don't know whether the build may use system regional settings, or that we have to set these regional settings ourselves somewhere and if so, I have no idea where or how.

ghtester, 2021-12-10:
OK, thank you for your detailed comment.  I also think it's somehow related with a region but if it's hidden setting somewhere in the WiFi library, I understand it'll be hard or impossible to solve... Despite AP channel 13 would be better in my environment, I changed it to 11 and I can live with that... Thanks again!

TD-er, 2021-12-10:
I'm now working on some other part of the code, but when I dive into WiFi again, I will keep it in mind.
Also found some more leads for improving WiFi scanning, so that part is on my virtual todo list (don't think I am so structured that I will have an actual todo list ;) )

TD-er, 2024-11-05:
OK, I know this is an old issue, but I've been digging into "hidden SSID" the last few days.

The reason why this is not working is this:
The default scan mode per channel is:

- active scan from 1 to 11
- passive scan from 12 to 14.

This can only be changed when the WiFi country policy is set to manual. (and channels > 11 are included in the policy)

So this will be addressed when I'm done with the (massive) WiFi/network refactor I am now working on.

I will also add some settings to store the preferred channel and/or BSSID to speed up connecting to hidden SSID networks.