letscontrolit/ESPEasy #3871: Missing PCF8575 support - please add (similar as plugin P019)

Cover1987, 2021-12-08:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

Checked the Max Build for ESP32

### Summarize of the problem/feature request

I have a Buttonboard that is using a PCF8575. The PCF8575 is not supported yet. Please add support for that I/O Expander.

### Expected behavior

Supports 16 I/Os

### Actual behavior

If i select PCF8574 (that one is recognized via I2C Scan even the bigger PCF8575 is build in) i get I/O Signal from PCF8575:
0x20 P0 from P00
0x20 P1 from P01
0x20 P2 from P02
0x20 P3 from P03
0x20 P4 from P04
0x20 P5 from P05
0x20 P6 from P06

P7 seems not to work.

i also get online response from espeasy at 0x3c (60) from P0 to P7 but no answer if i trigger ports of the PCF8575

### Steps to reproduce

1.  Expand the Plugin P019 within P8-P17 or
2.  Create a total new for the PCF8575
3. 

 yes

### System configuration

Hardware:
Used a ESP32-WROVER-B and a so called button baord from NERDIY.de
https://nerdiy.de/en/buttonboard/

ESP Easy version:  ESP_Easy_mega_20211105_test_A_ESP32-wrover-kit_4M316k Nov 5 2021

ESP Easy settings/screenshots: 

### Rules or log data

```

```

tonhuisman, 2021-12-08:
Related [Forum topic](https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=8893)

Have you tried the same test using the MCP23017 plugin, that does support 16 pins, and looks to be quite similar in software-interface?

Cover1987, 2021-12-09:
MCP23017 Test:
0x20 
PA0 nothing 
PA1 nothing
PA2 siganl from P01
PA3 siganl from P03
PA4 - PA7 nothing
PB0 signal prom P00
PB1 nothing
PB2 signal from P02
PB3 signal from P03
PB4-PB7 nothing

all other ports from other I2C Adresses are showen as offline

tonhuisman, 2022-04-18:
This could be partially solved (input only) via #3927

tonhuisman, 2022-04-22:
@Cover1987 the mentioned PR has been merged, does that help enough to close this issue?

Cover1987, 2022-05-11:
> @Cover1987 the mentioned PR has been merged, does that help enough to close this issue?

Seriously - I tried to understand whats going on on the P061 Merg, but i dont have a clue how to implement it

tonhuisman, 2022-05-12:
For the PCF8575 you will need a build that doesn't have LIMIT_BUILD_SIZE (compile-time option) set, and all TEST builds have that enabled...
You can optionally create a Custom build yourself (the standard Custom builds don't include this plugin), or...
I'll create a TEST_E build (or do you need some other specific plugin/build?), that does include the PCF8575, later today. Still at work...

Cover1987, 2022-05-12:
I've only the 4MB Version.
I'll give it a try tonight to compile it by myself. If i don't have success i'll come back to you <3

tonhuisman, 2022-05-12:
FYI: ESPEasy starter guide](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy) (and on the same page is the info for setting up VSCode)

Cover1987, 2022-05-14:
ehmm.. nope - i didnt get it working.
If You could compile it for me, i would be gratfull.
If possible with P019, P036, P045, P037. Thank you!

tonhuisman, 2022-05-15:
Here is a Custom build (based on the regular Custom configurations, with some plugins added)

[ESP_Easy_mega_20220515_custom_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/8694898/ESP_Easy_mega_20220515_custom_ESP32_4M316k.zip)

NB: Support for PCF8575 is _only_ added to the Keypad plugin ([P061](https://espeasy.readthedocs.io/en/latest/Plugin/P061.html#p061-page)), and _not_ to the Switch input - PCF8574! ([P019](https://espeasy.readthedocs.io/en/latest/Plugin/P019.html#p019-page))
The other plugins you requested are also enabled in this build.

Cover1987, 2022-05-15:
Thaks a lot.
I could flash it with out problems.
But i don't get most of the inputs to work,

If I select 9x8 Matrix i get following output from the logging:

`23197: Webserver args: 0: 'index' length: 1 1: 'page' length: 1
25675: Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 8 4: 'TDE' length: 2 5: 'i2c_ad
25679: SaveToFile: free stack: 6712
25812: FILE : Saved config.dat offset: 32768 size: 520
25813: SaveToFile: free stack after: 6712
25879: SaveToFile: free stack: 7080
26002: FILE : Saved config.dat offset: 32768 size: 520
26003: SaveToFile: free stack after: 7080
26052: SaveToFile: free stack: 7160
26560: FILE : Saved config.dat offset: 0 size: 3004
26560: SaveToFile: free stack after: 7160
26573: SaveToFile: free stack: 6600
26583: FILE : Saved security.dat offset: 1024 size: 128
26584: SaveToFile: free stack after: 6600
27123: KPad : ScanCode=0x0`

After that i get feedback by pushing buttons from:
P13 (Pin16) "KPad : ScanCode=0x4" and 0x0 if i release the button.
P12 (Pin15) and P11 (Pin14) "KPad : ScanCode=0x2" on both
P10 (Pin13) "KPad : ScanCode=0x1"

P00-P07 (Pin4-11) no feedback.
P14-P17 i dont use in my project.

Cover1987, 2022-05-15:
Using Direct16 i get following:

1027540: Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 8 4: 'TDE' length: 2 5: 'i2c_ad
1027559: SaveToFile: free stack: 6712
1027795: FILE : Saved config.dat offset: 32768 size: 520
1027795: SaveToFile: free stack after: 6712
1027869: SaveToFile: free stack: 7080
1028093: FILE : Saved config.dat offset: 32768 size: 520
1028094: SaveToFile: free stack after: 7080
1028154: SaveToFile: free stack: 7160
1028684: FILE : Saved config.dat offset: 0 size: 3004
1028685: SaveToFile: free stack after: 7160
1028699: SaveToFile: free stack: 6600
1028800: FILE : Saved security.dat offset: 1024 size: 128
1028800: SaveToFile: free stack after: 6600
1029007: WD : Uptime 17 ConnectFailures 0 FreeMem 221108 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
1029035: KPad : ScanCode=0x8 

Pushing and result: (after release i get 0x8
P00 (Pin4): ScanCode=0x1
P01 (Pin5): ScanCode=0x2
P02 (Pin6): ScanCode=0x3
P03 (Pin7): ScanCode=0x4
P04 (Pin8): ScanCode=0x5
P05 (Pin9): ScanCode=0x6
P06 (Pin10): ScanCode=0x7
P07 (Pin11): no feedback
P10 (Pin13): no feedback
P11 (Pin14): no feedback
P12 (Pin15): no feedback
P13 (Pin16): no feedback

tonhuisman, 2022-05-16:
Are pull-up resistors connected on **all** inputs of the PCF8575, as stated in the documentation? As this chip does _not_ have them internally, unlike the PCF857**4** which has these internal pull-ups.

Cover1987, 2022-05-16:
> Are pull-up resistors connected on **all** inputs of the PCF8575, as stated in the documentation? As this chip does _not_ have them internally, unlike the PCF857**4** which has these internal pull-ups.

Yes, every single input is own pull up resistor

closed 2023-09-02 by TD-er