letscontrolit/ESPEasy #3873: ESP crashes with {} in custom dashboard

V0JT48, 2021-12-11:
### Steps already tried...
- [ x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ x] Tested previous/other build (mention which one already tested)
- [ x] Tested on other node to make sure hardware isn't defective.
- [NA] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

Device reboots if custom dashboard source file contains {}

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

After upgrading to ESP_Easy_mega_20210802_normal_ESP8266_4M1M my custom dashboard stopped working and device resets when *.esp is accessed. Same results are with ESP_Easy_mega_20211105_normal_ESP8266_4M1M. It worked fine with ESP_Easy_mega-20200410_normal_ESP8266_4M1M and 20200801 release.
I tried to narrow down what part of my HTML code causes it. So far main issue seems to be related to curly braces but not all appearances. The simplies example that triggers the issue is below.
I have some javascript code in .esp files and this makes it impossible to use them.

### Steps to reproduce

Upload file test.esp and access it http://ESP_IP/test.esp
Content of test.esp is below.

Rebooting or powering off produces same result.

### System configuration
Rules and NTP enabled
Hardware: ESP12E

ESP Easy version: ESP_Easy_mega_20210802_normal_ESP8266_4M1M, ESP_Easy_mega_20211105_normal_ESP8266_4M1M

### Rules or log data

```
<!DOCTYPE html><meta name="viewport" content="width=width=device-width, initial-scale=1">
<HTML><HEAD><meta charset='utf-8'/></HEAD><BODY>
{}
</BODY></HTML>
```

TD-er, 2021-12-11:
Hmm I just took a quick look at the function to handle the custom page.
The main suspect so far seems to be the `parseTemplate` function.
But that would mean it would probably not only crash on parsing the dashboard file, but also lots of other places where this is called.
e.g. when used on a display (like Framed OLED) or in rules or generating log entries.

I will try to reproduce it here too with a debugger attached.

TD-er, 2021-12-11:
It seems like the ESP is then entering some infinite loop.
What kind of crash do you see? Does it actually crash, or can to press reset to get out of it?

V0JT48, 2021-12-11:
The deice reboots almost instantly. I have not tried it with serial port attached if some more details are visible.
After some experimenting possible workaround for javascrip is to remove all whitespace at begining of lines. Code is not so readable but page loads.

TD-er, 2021-12-11:
I tried your sample code on an ESP32 and it keeps "running" or at least is not rebooting.
The page seems to still be loading in the browser but noting seems to happen.

Also the ESP is no longer responsible until I reboot.

I think it may be a bug in `get_next_inner_bracket`, but still looking into it.

TD-er, 2021-12-11:
Yep the bug is in there.
Now have to find a proper work-around to make sure the {} is still removed.

V0JT48, 2021-12-11:
{} is just the simpliest example.
Crash:
```
if (x){
 a
} else {
 b
}
```
No crash:
```
if (x){
a
} else {
b
}
```

TD-er, 2021-12-11:
Can you test the fix I made here?
https://github.com/letscontrolit/ESPEasy/pull/3874

V0JT48, 2021-12-11:
Well it fixes the crash for {} but also removes it from HTML code which is also valid for evrything inbetween for cases it used to crash.
```
if (x){
 a
} else {
 b
}
```
Turns into
```
if (x) else 
```
Code stays with removed whitespace at begining of line. Hard to tell which option is better.

TD-er, 2021-12-11:
It is a simple fix to leave the {} in the text.
Will make it after I put my daughter to bed.

TD-er, 2021-12-11:
Just pushed a new commit which should be the best of both world:
- No crash
- Leave `{}` when it contains texts which cannot be matched to a command.

V0JT48, 2021-12-11:
It works great, thanks a lot for super fast fix.

TD-er, 2021-12-11:
Great!
Thanks for testing.

Will merge it now and it will be included in the next build. (hopefully tomorrow)

PR #3874 (2021-12-11): [dashboard] Fix crash on parsing empty command {} (#3873)
Fixes: #3873
commit 404b7aff (2021-12-11): [dashboard] Fix crash on parsing empty command {} (#3873)

Fixes: #3873
closed 2021-12-11 by TD-er