letscontrolit/ESPEasy #3875: Extend P1 Reader to support Swedish Version of HAN port

TD-er, 2021-12-11:
See [request on forum](https://www.letscontrolit.com/forum/viewtopic.php?p=56328#p56328)

Apparently the protocol differs.

TD-er, 2023-03-04:
Code example: https://github.com/romix123/P1-wifi-gateway/blob/8827422c9ac9aab94b45c4e6f58495c4b0d04a00/src/decoder.ino#L192

N.B. this code example is very verbose and can use quite a lot of optimizations regarding telegram decoding.
For example there is no need to compare with strings over and over again, when we simply can decode the telegram header into various ints

tonhuisman, 2023-03-04:
An alternative could be this template-based DSMR P1 parser: https://github.com/lvzon/dsmr-p1-parser, though that doesn't support the Swedish version of the protocol, it wouldn't be hard to add.