letscontrolit/ESPEasy #3879: TaskRun behaviour

chemmex, 2021-12-14:
Clean 20211105_normal image, BME280 and PMS7003 installed. 

It seems that task values of P028 and P053 are read only by the interval set in the task menu; TaskRun command shows only "Ok" output but no values, both via web command and being invoked from rules. No log records follow, mqtt sends out still values from rules.
At the same time, P026 Sysinfo plugin provides output regardless of interval setting.

TD-er, 2021-12-14:
The BME280 plugin for sure, and I guess the PMS x003 may also be using the same strategy.
The BME280 does first initiate a read when PLUGIN_READ is called and it re-schedules itself when the new values should be ready.

This means the new values are not there yet, right after calling taskrun.
But there will be a new event as soon as there are new values from that task.

The PMS x003 does have a slightly more complex situation regarding taskread.
The first N seconds after starting the sensor (e.g. when waking from deep sleep) the values are ignored as those are not reliable (stated in the datasheet)
I think N is 20 sec here, but might be 30.
The logs do show whether a new value is suppressed because of this.

chemmex, 2021-12-14:
Yes, I'm aware of warm-up time for PMSx003, I use to control it through the SET line via rules and call TaskRun after 30 seconds. Finally, it seems to work as expected.

But for BME280, what strategy can be used to take readings manually? I wish to avoid using regular interval since it is usually variable.

TD-er, 2021-12-14:
If you call taskrun, then it will after some time (1700 msec from the top of my head) call itself again to actually send the new data to the attached controllers and send events.
Thus calling taskrun on the BME280 will exactly do that, but it will only yield new values after some time.
So you don't need to run it at a regular interval, but can trigger reads yourself.

If for example you have the interval set to every 600 sec, then it will reschedule itself at 600 sec when you called a taskrun.
But if you called the taskrun command more frequently it will give a new sample at the intervals you trigger the taskrun command.

chemmex, 2021-12-14:
Tried to follow the path you described, but no luck so far. 

Below is a piece of log after I re-enabled BME280 task with 100 sec interval. Time events are omitted.

```
172729: BMx280 : Detected BME280
172738: EVENT: TaskInit#BME=1,1
176222: BME280: dew point 5.13C
176233: BME280 : Address: 0x76
176234: BME280 : Temperature: 24.29
176235: BME280 : Humidity: 29.05
176236: BME280 : Barometric Pressure: 996.16
176241: EVENT: BME#Temperature=24.29
176338: EVENT: BME#Humidity=29.05
176438: EVENT: BME#Pressure=996.16
202265: HTTP: taskrun,1
202266: Command: taskrun
280222: BME280: dew point 6.64C
280233: BME280 : Address: 0x76
280234: BME280 : Temperature: 23.90
280235: BME280 : Humidity: 33.03
280236: BME280 : Barometric Pressure: 996.15
280301: EVENT: BME#Temperature=23.90
280401: EVENT: BME#Humidity=33.03
280501: EVENT: BME#Pressure=996.15
```

As you may see, TaskRun hasn't triggered an event, and next values has come right after the regular interval. 
I also tried to catch the values with mqtt publish, but they are frozen at the initial reading.

TD-er, 2021-12-14:
Hmm have to look into it, as it does look like a bug after all...

TD-er, 2022-06-17:
OK, totally forgot about this one...
Thanks for bringing it to my attention again.

PR #4100 (2022-06-23): [BME280] Allow to take samples using taskRun (#3879)
Fixes: #3879
commit 92cd71bb (2022-06-23): [BME280] Allow to take samples using taskRun (#3879)

Fixes: #3879
closed 2022-06-23 by TD-er