letscontrolit/ESPEasy #3881: Plugin_096 - eInk Display 1.54 Waveshare with Waveshare ESP8266 Driver Board issues

NorbertRoller, 2021-12-15:
Hi, I do need your help, as I got stuck since a few days. Out of frustration I ordered the original Waveshare hardware, but the result is the same. (Compared to my own build ESP8266 Setup and the NodeMCU)

I had loaded the Wavshare Software to test the Controller and the display and it works without any issues. Pictures can be uploaded and the screen is fully refreshed.

Now I installed mega-20211105, Display 4M1M
SPI is on, I2C is off,
The PIN definition was updated to match the source code of Waveshare
SDI-GPIO13
SCLK-GPIO14
CS-GPIO16
D/C-GPIO15
Rst-GPIO2
Busy-GPIO5

I can send commands and see the display flashing. But it only clears 2/3 and whatever I write or draw doesn't show at all.

I checked the code and it almost looks like the width and hight of the configured device (200x200) is not updated correctly and the default vaues are used.

Plugin_096_EPD_SettingStruct()
  : address_epd_cs(EPD_CS), address_epd_dc(EPD_DC), address_epd_rst(EPD_RST), address_epd_busy(EPD_BUSY), rotation(0), **width(250), height(122)**

Also every few commands the GUI stalls and I either have to press reset or refresh to the Main Page (default IP).
It might be a connected issue, as the width is to large and the controller may not respond properly.

This is the board:
https://www.waveshare.com/e-paper-esp8266-driver-board.htm
This is the B/W Display
https://www.waveshare.com/product/displays/e-paper/epaper-3/1.54inch-e-paper.htm

Many thanks in advance

tonhuisman, 2021-12-15:
Ah, those eInk driver chips are quite similar, but seem to have small differences, making them all incompatible.

I've already added 2 different controller chips to this plugin, in pending PR #3761, and that also allows to configure the width and height of the display (and a _few_ more features 😋), but that most likely won't (yet) help to get it going. You can try that by downloading that from the Github Actions page, [here](https://github.com/letscontrolit/ESPEasy/actions/runs/1570296770) (get the Binaries artifact and extract the ESP8266_DISPLAY bin file).

If you could track down the actual controller chip involved, it would help to add support for this display. (I already have another Waveshare eInk display waiting on my workbench but didn't get to implementing support yet)

NorbertRoller, 2021-12-16:
Hi, non of the three abailable chipset work at all.
I have searched a lot and did only find one hint so far. The chipset is supposed to be a IL3829.
The display supports flicker free partial update. Because of that the V2 seems to have a different code in most of the libraries.
On the waveshare page (link above) there is a demo programm that has .h files for all their displays. It shows init and write sequences that might help and speed up your development.

About the implementation in ESPEasy: On the Device overview page (with the 12 lines) the parameter are shown in Column GPIO. Just for D/S and Reset it shows the GPIO only. So the description is missing. No a big thing, but I run into it during debugging.

Do you have any idea when a driver might be available?
I have to decide if I wait or change to an 1.54 LCD/OLED screen.

NorbertRoller, 2021-12-16:
This is the init sequence from Waveshare code for the V2 (V1 ist very different)
EPD_SendCommand(0x01); //Driver output control
        EPD_SendData(0xC7);
        EPD_SendData(0x00);
        EPD_SendData(0x01);

        EPD_SendCommand(0x11); //data entry mode
        EPD_SendData(0x01);

        EPD_SendCommand(0x44); //set Ram-X address start/end position
        EPD_SendData(0x00);
        EPD_SendData(0x18);    //0x0C-->(18+1)*8=200

        EPD_SendCommand(0x45); //set Ram-Y address start/end position
        EPD_SendData(0xC7);   //0xC7-->(199+1)=200
        EPD_SendData(0x00);
       EPD_SendData(0x00);
        EPD_SendData(0x00);

        EPD_SendCommand(0x3C); //BorderWavefrom
        EPD_SendData(0x01);

        EPD_SendCommand(0x18);
        EPD_SendData(0x80);

        EPD_SendCommand(0x22); // //Load Temperature and waveform setting.
        EPD_SendData(0XB1);
        EPD_SendCommand(0x20);

        EPD_SendCommand(0x4E);   // set RAM x address count to 0;
        EPD_SendData(0x00);
        EPD_SendCommand(0x4F);   // set RAM y address count to 0X199;
        EPD_SendData(0xC7);
        EPD_SendData(0x00);
        while (digitalRead(BUSY_PIN) == 1) delay(100);
        Serial.print("\r\n init over");

TD-er, 2021-12-16:
> while (digitalRead(BUSY_PIN) == 1) delay(100);

That's for sure a line that will **not** be copy/paste into ESPEasy code.

NorbertRoller, 2021-12-16:
If the Busy Pin is not connected / configured, you might not have another option.

TD-er, 2021-12-16:
> If the Busy Pin is not connected / configured, you might not have another option.

Sure there is.
This is just halting the entire ESP so nothing will work anymore if the pin is not making proper contact.
Just schedule some new call and mark the current state to be waiting for the pin to be released.
ESPEasy will never have potentially never ending busy wait loops as there are way more elegant ways to solve this.

tonhuisman, 2021-12-16:
> Do you have any idea when a driver might be available?
> I have to decide if I wait or change to an 1.54 LCD/OLED screen.

That Waveshare repository is quite helpful, so I'll be fetching initializations from there.

If you go for minimal power-use (and infrequent display updates) then an eInk screen is quite efficient, if you want immediate or (very) frequent screen-updates, then OLed of TFT would be a better choice. For OLed currently only SSD1306 and SH1106 are supported, and the largest with these chips I've seen are 1.3". (I have some more of these displays on my wish-list, but that I'll probably pick up later)
For TFT I've added quite some options, resolutions and sizes in the PR mentioned earlier.

> > while (digitalRead(BUSY_PIN) == 1) delay(100);
> 
> That's for sure a line that will **not** be copy/paste into ESPEasy code.

> If the Busy Pin is not connected / configured, you might not have another option.

The delay will be much shorter, if any, but these eInk display are already quite demanding in the memory department, but it depends on the selected eInk controller. Maybe an ESP32 is a better unit for this (that's also supported by ESPEasy).

TD-er, 2021-12-16:
I guess there will be no delay.
If one needs to wait for the device to process stuff, just set the current state and try again in the next loop.

NorbertRoller, 2021-12-16:
My project is one screen update per minute, only.
It will show date + Time (hh:mm) and Temp and Humid.
So nothing critical with a lot of sleep inbetween

TD-er, 2021-12-16:
> My project is one screen update per minute, only. It will show date + Time (hh:mm) and Temp and Humid. So nothing critical with a lot of sleep inbetween

Yep, but all other users that may use it will run into these issues.
So better make it right in the first place and that's what we're doing with the ESPEasy code.
People are even running out of the 32 task limit on ESPEasy running on ESP32, so you can be sure there will be use cases where it will be used to the max.

NorbertRoller, 2021-12-16:
Agreed.
I might run out of Pin's as the sensor is I²C and I need one more for a switch.
So I may end up on using ESP32, too.

tonhuisman, 2021-12-16:
> My project is one screen update per minute, only.

For an eInk that is quite often/fast, as each screen update can take between 6 and 12 seconds...

NorbertRoller, 2021-12-16:
My idea is to load the template once per hour and after that only update the 4 sections once a minute.
I will aim for a layout that has less regions to update. So I might end up with 1 or 2 very small regions.

tonhuisman, 2021-12-16:
The current plugins, both the 'old' or the 'new' as PRed, do not support partial updates, AFAIK. Even though the Lolin driver does have support for that.
I'll see if I can implement support for that in the PR (for the displays that support that).

TD-er, 2021-12-16:
Just be careful not to update pixels too often on an e-ink screen.
The pigment literally wears off from the pixels. The pixel of an e-ink physically rotates inside some container and it does have some pigment on one side which wears off when rotating the pixel.
An update per second will wear out a screen in a weekend as the e-ink of a friend of mine showed. (an error in his programming)

NorbertRoller, 2021-12-16:
Very valuable information from both of you.
I am working at the layout and if I have to update the whole screen at once this is good to know. 
Can I send any pics or just characters with the plugin?

NorbertRoller, 2021-12-24:
Info have an ESP32 Development Board available for testing, now.
Let me know of I can help testing.

NorbertRoller, 2022-02-05:
Even so there was no answer, I have programmed a solution, myself.
I used the Waveshare ESP32 - EPD Modul and some free time to code the function myself. Waveshare offers some good samples and it wasn't too complicated to add the few extra functions.
The Project was to send a command to the Ventilation system via FHEM once a touch button is pressed. In this case show a picture and once finished, show Time, Temperature and Humid of the Bathroom. Temp and Humid are I²C connected.
The complicated part was to setup the EPD so parts of the screen can be updated without flickering and a refresh of 1 sec.
Below are some pictures.
[image]
[image]

closed 2022-02-05 by NorbertRoller

tonhuisman, 2022-02-05:
It would be appreciated very much if you would share your code, so it can be integrated in ESPEasy, and benefit others too.
This can be done by opening a Pull request, or dropping a zip file with the changed sources in this thread, whatever is most convenient for you.
TIA.