letscontrolit/ESPEasy #3882: read serial data?

hoangtntx, 2021-12-20:
I have a esp8266 device that send data and control through serial. how can i receive this data to my esp32 installed espeasy. I use serial server but cant parse data to value, only see it when telnet to tcp port. I tried serial proxy but dont know how to work with it. Anyone has solution. thx

oiiooiiooiio, 2023-01-22:
#3980

tonhuisman, 2023-01-22:
> #3980

Your PR doesn't add anything related to the question asked.
Adding a link to either the [Serial Server](https://espeasy.readthedocs.io/en/latest/Plugin/P020.html#p020-page)/[Ser2Net wiki page](https://www.letscontrolit.com/wiki/index.php?title=Ser2Net) or [Serial Proxy](https://espeasy.readthedocs.io/en/latest/Plugin/P087.html#p087-page) plugins documentation (though rather minimal) seems more helpful.

tonhuisman, 2023-09-09:
This can be closed, as there is no more response.

closed 2023-09-09 by TD-er