letscontrolit/ESPEasy #3883: CSS changes proposal, (PR would follow)

soif, 2021-12-21:
Hi there

I've slightly modified the EspEasy CSS (see following screenshots).I tried to make it a little more clean , and less 90's like, while respecting your current graphic identity, and keeping the CSS size as small as possible.

Here is what I've currently done.
- add more padding/margin  in header and footer and between blocks
- fix the tabs icons and sizes
- lighten the blue of the even rows in tables
- Start to fix the footer (I will need to remove some unwanted tags from the C code, first)
- slighty change the Tables headers, as well as H3s titles
- lighten the Notes
- Changed the info and help buttons (would really like to replace them by some FontAwesome icons)
- added more space between button and desc in the tools page
- lighten the Inputs background

Questions:
1) Do you like it? Want me to submit a PR?

if yes:

2) I woud certainly remove some tags/styles from the footer's element, which should slighty save some precious memory, while making it more style-able. OK?

3) I noticed that the CSS and JS are now offloaded from a remote server. Will it be the new way of serving medias ressources (means a working internet connection would now be required, every time)?
If yes, could I really increase the CSS file size, and maybe also add some external fonts, Ie adding [FontAwesome icons ](https://fontawesome.com/v4.7/icons/) or even images (ie EspEasy Logo)?

4) Do you only need the espeasy_default.css file or you want me to also submit the minified version?

Happy Christmas! 

[image]

[image]

[image]

[image]

TD-er, 2021-12-21:
It does look more clean.
Have to check it on a mobile too to see if things may still work as expected.

tonhuisman, 2021-12-21:
You might also have a look at the Dark Mode proposal that's already PRed in #3806 (and a few more similar PRs by the same author) and possibly combine/merge your efforts?

soif, 2021-12-21:
@TD-er Happy that you like it.

Regarding Mobile, I've not made extensive tests, but as I did change only minor things like padding and colors, I guess that It wont be really different.

Here is an example on width a smaller width:

[image]

----

Regarding question 3: Can I enlarge the css, or try to keep it small? Add externals fonts/images ?

Then I'm gonna push a PR so that you can inspect it...

tonhuisman, 2021-12-21:
Currently both the regular and minified files are there, so a PR should include both, IMHO, as they should be functionally the same.

soif, 2021-12-21:
@tonhuisman 
Regarding Dark Mode proposal in #3806. I might merge it but you'd have to decide if:
- we now default to dark mode only
- or light mode is default, and there is a JS switch to change to Dark Mode
- or Dark mode is a separate, optional css to upload

tonhuisman, 2021-12-21:
Dark mode is determined via a media query, so would be a user-selected system setting, no need to add any JS or option for that, I'd expect.

TD-er, 2021-12-21:
Ideally we should switch between light/dark mode using settings. So we can serve part of it by serving some basics dynamically (e.g. some color defines, used in the CSS)

I would like to have the CSS as small as possible so we can store it in the build, or load from the file system.
Serving from an external host is a last resort for space restricted builds.
Maybe we can also add a setting to save the URL to load the CSS from an external host (or another node?)

soif, 2021-12-21:
I've pushed a PR....

@TD-er : switching between css, via a setting, would be cool, but adding this would obviously expand the flash size both for the setting form and sentences, and by making the CSS way larger (ie certainly at least  1/4 more size).
Is this REALLY needed ?

soif, 2021-12-21:
BTW is there a script, or do you know an online service that can quickly, in one way, convert from string (ie CSS) to the PROGMEM hex array?

TD-er, 2021-12-21:
> BTW is there a script, or do you know an online service that can quickly, in one way, convert from string (ie CSS) to the PROGMEM hex array?

See the `embed_files.sh` script.

soif, 2021-12-22:
exactly what I needed! 💪  

thanks 🍺

closed 2021-12-23 by soif

tonhuisman, 2021-12-23:
By properly referencing your issues from PR's, (f.e.: resolves #issuenumber) the issue is closed automagically by Github, [see here](https://stackoverflow.com/questions/60027222/github-how-can-i-close-the-two-issues-with-commit-message), so you don't have to do that manually, and other people don't get the impression it is already solved, while the PR isn't merged yet.

soif, 2021-12-23:
thank you very much for this information!

 I was not aware that _"the issue is closed automagically by Github"_, and I just wanted to help by closing this issue (saving your time), that is now continuing in the corresponding PR....