letscontrolit/ESPEasy #3887: ESP32 + Esp_Easy Login after setting password not possible

m50434, 2021-12-25:
Hi,
I got an ESP32 POE from Olimex and flashed ESPEasy the way its explained here: https://github.com/letscontrolit/ESPEasy/issues/2384

The flash works. I used  ESP_Easy_mega_20211224_test_A_ESP32_4M316k-factory

But after I set a password, I cannot login to the ESP anymore.
I tried it with a simple password (abcd). I used "admin" as username and my setted password, but I cannot login anymore.
I tried it a few times (because first I thought I typed my password wrong).

I also tried it with some other firmwares:  ESP_Easy_mega_20211224_normal_ESP32_4M316k_ETH (which is the one I want to use) or ESP_Easy_mega_20211224_test_A_ESP32_4M316k_ETH-factory but everytime the same: After setting a password, I cannot login anymore....

Maybe there is a bug in this particular ESP32-Firmware?

m50434, 2021-12-25:
Me Again... Sorry,
Just realized its the same problem like here: https://github.com/letscontrolit/ESPEasy/issues/3860

With Internet Explorer I can still login. With Chrome, Forefox or my mobile chrome it does not work.

TD-er, 2021-12-25:
Since this one is indeed a duplicate of the one you mentioned, #3860 , I will close this one.

And I still have no clue what may be the problem here, as I can use it on my test nodes without any issues.

closed 2021-12-25 by TD-er

arekmitura, 2024-04-19:
Maybe this topic is old but I had same problem and looking for solution I found it and I have some conclusion I would share.

I can suggest that reflashing the ESPEasy didnt solve this because the config are stored in memory that is not erased during flash. So after flashing you can see that ESPEasy works like before I mean for example its try to connect to the same Wifi network like it was before flashing etc you can see it by serial monitor logs. 
Maybe there is some way to repair it by change AP password by serial port but I choosed hard way so How to deal with that fast but in hard way?
1. Open serial communication with ESP for example by Arduino IDE>tools>serial port monitor (remember to config right communication speed with ESP, I prefer that way because you see realtime logging) than you can see logs what currently ESP doing and there u can use commandline to type commands from ESPEasy documentation: [ESP commands](https://espeasy.readthedocs.io/en/latest/Reference/Command.html)
2. Now you can use "Reset" command it will clear all config and set ESPEasy to factory settings and voilà, now you can connect to AP ESPeasy by its default AP password but unfortunately you lost all configs so if someone have something big to do with that device I suggest to always make atleast 2 config backups before changes and after changes

TD-er, 2024-04-19:
If you still know the password, you can also try `clearpassword,<yourpassword>` from the serial console.

N.B. on more recent builds, if you call `reset` (which I still need to change to `factoryreset` as it can be mistaken for reboot), the ESP will still know about its WiFi credentials as those are also kept in NVS storage (on ESP32).
The admin password is not kept in NVS storage for exactly this reason.
All which is checked on the "Factory Reset" page will be kept in NVS and re-used when performing a factory reset.

uzi18, 2024-04-20:
@arekmitura Do you have backup of file security.dat ?

tonhuisman, 2024-05-02:
> I can't find way to change STA MAC address of ESPEasy, do you know maybe how to change it? Could you help me, please?

There is no configuration option available in ESPEasy for changing the MAC addresses, it's using the default MAC addresses of the ESP, both STA and AP.

NB: This support question should best have been asked in [the ESPEasy forum](https://letscontrolit.com/forum/index.php), not in an already closed support issue...

TD-er, 2024-05-02:
What usecase do you have to change the MAC address?