letscontrolit/ESPEasy #3894: SDS011 not working possibly due missing firmware state reset

ikkefc3, 2021-12-30:
Hi all,
I have a problem getting readings from my SDS011. The sensor just seems completely dead. It sometimes spins up, but ESP Easy is not picking up any data. I also confirmed that the same issue is happening on another SDS011 I have.
I have had this issue since I used the SDS011 and ESP easy, since the beginning of this year. note that I am using the latest ESP easy TEST-A (mega-20211224).

The issue starts when I turn off the ESP32 unexpectedly (power loss for example). It looks like the firmware of de SDS011 is in the wrong state for getting a reading. I tried resetting the sleep time to 0, but that did not make a difference. What does seem to be working, is connecting the SDS011 to my PC and running the test script from this python library on my PC:
https://gitlab.com/frankrich/sds011_particle_sensor/-/tree/master/Code

In particular the reset() command of the library, which sets Report mode to Initiative, workstate to measuring and duty cyle to 0.
For me it would be really helpful to have an option in the ESP Easy interface to do this reset, so I do not have to disconnect the sensor and run it on my pc every time this happens.

Thank you for reading.

fredvduin, 2022-01-11:
I had an issue with sleeping mode, after changing the interval to 300 sleep time works.
I change the interval afterward to 60 and it is still working.
Probably it takes more than 60 seconds for initializing (firmware/config?)
Used download Binary ESP_Easy_mega_20210503_normal_ESP8266_4M1M

ikkefc3, 2022-01-11:
Thank you for your suggestion.
I tried running it for a day to see if I was being to impatient and I tried different sleep mode values but it still didn’t detect it. It’s as if the ESP listens for data at a certain time and when the firmware and esp don’t sync up it never works.

fredvduin, 2022-01-11:
Has you changed the interval?

This is my config
[image]

ikkefc3, 2022-01-11:
Yes, I tried 1, 30, 60 and 600 seconds as a value.

TD-er, 2022-01-11:
Is the RX also connected on the SDS unit?
And I see you're using software serial.
Which pins do you use?