letscontrolit/ESPEasy #3895: [PluginPlayground]: Migrate NeoPixelBusFX to ESPEasy mega branch

tonhuisman, 2021-12-30:
On a remark of @clumsy-stefan as it has been stable for a very long time.

tonhuisman, 2021-12-31:
@clumsy-stefan 
I've reserved Plugin-ID 128 / `USES_P128` for this migration.
As the current ESPEasyPluginPlayground `_P124_NeoPixelBusFX.ino` source doesn't compile on latest ESPEasy sources (after adding the required libraries, and minimal required 'corrections'), I assume you have a patched version available already. Can you either create a PR with that, or attach it here so I can use that to create a working plugin & PR?

clumsy-stefan, 2021-12-31:
No, I use the latest version from the playground which compiles with latest version from ESPEasy... What problem does it report?
Actually I guess the issue is I don't use the version from PluginPlayground but from
https://github.com/djcysmic/NeopixelBusFX
Sorry, I forgot that...
Probably that solves the issues?

PR #3899 (2022-06-15): [P128] Migrate NeoPixelBusFX from playground / djcysmic/NeopixelBusFX
Resolves #3895 
Resolves #1971

- Add plugin NeoPixelBusFX, migrated from ESPEasyPluginPlayground and djcysmic/NeopixelBusFX
- Enable support for ESP32/ESP32s2

Available in (new) `NEOPIXEL` builds.

Known issues:
- ESP8266: H/W is fixed using GPIO-2 (Library only supports this GPIO)
- ~ESP32: H/W is fixed on HSPI~ (Fixed: Switched to Rmt method, that allows GPIO selection)
- Only supports GRB-mode, other modes require re-compilation
- Only supports NeoPixel LED-strips, not DotStar LED-strips.

TODO:
- [x] Select build set t [... cut]
closed 2022-06-15 by TD-er