letscontrolit/ESPEasy #3900: P1 data on wrong place in MQTT-message

georgeavd, 2022-01-05:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [X] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [X] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

The P1-topics are published with and empty placeholder and the status header is missing.

### Expected behavior

The P1-topics should be presented with a correct label (E.g. P1) underneath the same label (hostname) as the status (LWT-topic) messages.

### Actual behavior

The data is currently not published according to the default settings (see screensot below) 
[image]

### Steps to reproduce

1. Check with E.g. MQTT-Explorer

The problem persists after powering off and on

### System configuration

Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20211224_f162ebf
Plugin Count:⋄
47 [Normal]
Build Time:⋄
Dec 24 2021 14:39:35
Binary Filename:⋄
ESP_Easy_mega_20211224_normal_ESP8266_4M1M
Build Platform:⋄
Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20211224_f162ebf

Hardware: Wemos D1 with P1-shield
ESP Easy version: Build: ESP_Easy_mega_20211224_normal_ESP8266_4M1M Dec 24 2021

ESP Easy settings/screenshots: 
[image]

### Rules or log data

TD-er, 2022-01-05:
What format do the P1 messages have?
That plugin is relatively old, so it does not support string types as that was introduced later. (and not fully supported yet)

What kind of data does the task now output (e.g. screenshot from the devices page) and what do you expect?

georgeavd, 2022-01-05:
The tcp-output in port 8088 is as expected, but you're right there are no values presented in the device table.
[image]

My Kamstrup is presenting data according the DSMR V2.2 specifications.