letscontrolit/ESPEasy #3901: MAX7219 decimal place shows wrong numbers "7dt & 7ddt"

AVRnet, 2022-01-05:
Hi, my Display show wrong numbers on decimal place with both commands 7dt and 7ddt.

http://172.17.2.20/control?cmd=7dt,10.2

2722266: HTTP: 7dt,10.2
2722268: 7DGT : Show Temperature=10.20
Display shows on decimal place 10.1

likewise commands with wrong numbers:
"?cmd=7dt,10.4" shows 10.3
"?cmd=7dt,10.7" shows 10.6
"?cmd=7dt,10.9" shows 10.8

The following shows correct numbers on decimal place:
"?cmd=7dt,10.0" shows 10.0
"?cmd=7dt,10.1" shows 10.1
"?cmd=7dt,10.3" shows 10.3
"?cmd=7dt,10.5" shows 10.5
"?cmd=7dt,10.6" shows 10.6
"?cmd=7dt,10.8" shows 10.8

Same problem with the 7ddt command.

I have tested diverse Generic ESP8266 and Generic ESP32 Boards with actual release "ESPEasy-mega-20211224.zip"
and use Atomic with PlatformIO to compile the sources.

tonhuisman, 2022-01-06:
I guess it has to do with these lines of code: https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P073_7DGT.ino#L1248-L1250
I'll have a look at correcting this.

tonhuisman, 2022-01-06:
I've created a pull request for fixing the rounding issue.
PR build will be available on this [Github Actions page](https://github.com/letscontrolit/ESPEasy/actions/runs/1664657915) (download the `Binaries.zip` once it's available)

AVRnet, 2022-01-06:
Great, all works fine now.
Many thanks for your support and the very quick correction!

tonhuisman, 2022-01-06:
Please re-open, it will be closed automatically when the PR is merged (because I linked it in the PR. I can't merge it myself, that's for the repo-owner 😉), and others can see this is still a work in progress.

AVRnet, 2022-01-06:
Ok, i´m a newbie on github :)

PR #3902 (2022-01-07): [P073] Fix rounding errors on 7dt and 7ddt commands
Resolves #3901 

Correction in calculating the temperature values for display, as they are multiplied by 10, but the rounding was handled wrong.

TODO:
- [x] Testing (confirmed [here](https://github.com/letscontrolit/ESPEasy/issues/3901#issuecomment-1006921946))
closed 2022-01-07 by TD-er