letscontrolit/ESPEasy #3904: Where find last version ROM Normal 8266_4096

sigalou, 2022-01-10:
Hello,

I'm looking to find the most recent 4096k version of Easy_Mega.
This version has disappeared from ZIP files.
Do you know how long I have to go up to find the latest version that fits 4096k of memory?
thank you very much

tonhuisman, 2022-01-10:
AFAIK, that has been upgraded to include the size of the filesystem, so you are probably looking for a 4M1M build.
What is the entire name of the .bin file you have used before? As currently there are quite some more builds for ESP8266, 'normal' should match the regular build, and any of the 'test' (A..D) should include all plugins that are in previous 'test' builds, and also include the 'normal' or stable plugins.
The current list of available plugins, and the build they are in, can be [found here](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html)

sigalou, 2022-01-10:
ohhhhhh 
Sorry, I was wrong, it's 4096 and not 4096k
So I am looking for the latest version on the format normal_ESP8266_4096.bin
Thank

TD-er, 2022-01-10:
The "4096" notation was from years ago.
I renamed that to "4M1M" as it describes a 4M flash chip.
The partitioning used back then was using an 1MB partition for storing the file system + settings.
So to upgrade that board, you can use a build with "ESP8266" and "4M1M" in the name.

sigalou, 2022-01-10:
Thank you very much !!!!
Just a name change then. I was looking for 4096 because in some (old) documentations it is noted like this.
Thanks thanks !! You can close or delete my ridiculous question

closed 2022-01-10 by sigalou