letscontrolit/ESPEasy #3910: USE_Servo missing in ESP_Easy_mega_20211224_normal_ESP8266_4M1M.bin

Chris87-ux, 2022-01-12:
Hello,

I flashed ESP_Easy_mega_20211224_normal_ESP8266_4M1M.bin on my WEMOS D1 mini to do some tests on the HX711, but as I tried to also connect a Servo I get this error:
[image]
 According to the documentation, this USE_Servo should be a part of every build.

Br Chris

tonhuisman, 2022-01-12:
Are you sure you have the `NORMAL` build running here? The USE_SERVO feature is known to be disabled for the `TEST` builds and other size-challenged builds, but should in fact be available for `NORMAL`. (Checked the build config, not an actual build, though.)
The HX711 is only available in the TEST_A builds, so having that in your log is a bit of a giveaway...

tonhuisman, 2023-09-01:
No follow-up, but AFAICS can be closed.

closed 2023-09-01 by TD-er