letscontrolit/ESPEasy #3919: Crashing when MQTT controller is in use while the connection is unstable

ghtester, 2022-01-19:
I am still facing to ESP node crashes related with MQTT & unstable connectivity.
The MQTT controller is used (the broker is located remotely in public Cloud) to receive data from sensors every few seconds (for test purposes, to reproduce the issue more quickly).
I have also tested the behaviour with local MQTT broker with the same or similar results.
As soon as there are some connection interruptions (when the WiFi connectivity is lost or just the MQTT broker becomes unavailable ), it leads to ESP node crashes. It seems it's due to memory exhausting, despite the MQTT controller configuration should discard the data that can't be sent to broker which is unavailable.
When I disable the MQTT controller, the node stops crashing.
Latest custom Vagrant build was dated 20220111 but I am observing this issue for months. 
The configuration screenshots:
[image]
[image]
[image]
[image]

TD-er, 2022-01-19:
> (the broker is located remotely in public Cloud)

If so, then please increase the timeout to about 1000 msec.

Can you show the last few items on the tools->Advanced page?
If periodical scan WiFi is enabled, disable it.
The opposite for always using max. TX power, set it to enabled.

ghtester, 2022-01-19:
Thanks for the hints, I have already tried to increase timeout (between 500-800 ms for sure) but I did not encounter any positive feedback and as mentioned, the same issue happens even with local MQTT broker (on local LAN).
The periodical WiFi is permanently disabled. I can play with WiFi settings but I don't expect from WiFi to be permanently connected, there will aways be ocassional WiFi connectivity drops and I believe ESP node should not crash in such cases.
The HTTP controller does not make such issue, it's related just to MQTT.
[image]

TD-er, 2022-01-20:
Please enable Send with Max TX power
And if it is still available in your build, disable the periodical scan WiFi.

ghtester, 2022-01-20:
Well, what I tried to mention - although the WiFi connection is stable, the crash issue also happens when just the MQTT broker is unavailable. So it is unlikely the WiFi power could help. I believe I tried evey possible parameter in past but nothing helped so far.

TD-er, 2022-01-20:
Let me explain what I expect is happening here.
I do set the timeout on the network client when opening a connection to another host.
However, there are some (hard to reproduce) situations where this timeout is not being honored.
This would then result in a watchdog reset.

To why making a connection may timeout on a sub-optimal connection, there are lots of options.
- TX power is too low, so access point may not always receive the ESP's request. => set max. TX power
- Access point (or other switches in the network) may not know how to reach the ESP for a reply/ack/etc. => Gratuituous ARP
- Lots of other traffic/noise/etc. => Force WiFi B/G to allow for slightly more signal/noise margin.
- Multiple access points with the same SSID/pass => Don't do that for now to test if it may help here.
- No fixed channel on the access point => Don't do that for now to test if it may help here.

If the reboot reason does not have anything to do with the watchdog, but more like an exception, then it might be your ESP is running out of memory or your MQTT messages are rather long. Try tweaking with the controller settings.

tonhuisman, 2022-01-20:
The screenshot shows you are on a 20211202 build of ESPEasy, upgrading to the *20211224* build could help here, as some possibly obscure bugs seem to have been resolved.

TD-er, 2022-01-20:
Well I know for sure there is a bug in checking if the device is actually connected.
And that's what I have been working on for the past ~3 year as it is next to impossible to rely on the connected state as reported by the SDK.

ghtester, 2022-01-21:
I understand the complexity of WiFi connectivity layer and that due to bugs in SDK you may have a false info about connection status. I also understant it may lead to watchdog activation when it's not possible to recover properly from connection timeouts.
 But I don't get why there are (in most cases) Exceptions due to exhausted memory when the MQTT controller has an issue with connection reliability to MQTT broker. In my test environment I am sending the MQTT data every second (to local MQTT broker on LAN, not to remote one in public cloud). So it looks to me the bug could be in MQTT queue management. The strange thing was - when I turned off the local MQTT broker and rebooted ESP node (with enabled MQTT controller), it usually started generating Exceptions after some time from boot (and rebooted periodically). But not always, sometimes the ESP node could work for many days (with MQTT connection status permanently RED).

TD-er, 2022-01-21:
My first guess would not be the queue management, as it has been extremely thoroughly tested.
Sure there are numerous reports of MQTT + crashes, but just because of those numerous reports, that section has been looked into extensively.

I am thinking more in the direction of the plugins you're using, which I have not tested myself.
For example you seem to be running the LCD plugin.
I have not yet looked at that code, but let's assume that plugin is allocating for whatever reason 2k of memory on each update.
The queue management of the controllers does have a memory check which is limited at 3k free. Thus if the free memory gets below this threshold, it will no longer accept new messages to add to the queue.
If the MQTT controller is disabled, there is no queue to be filled and thus there is no significant drop in free memory possible on a WiFi disconnect.
When the low memory protection of the queue management kicks in, the free memory is somewhere around 3k, but that doesn't take memory fragmentation into account.
If one of the other plugins then needs to allocate a larger chunk of RAM, it is possible that the largest free block is only a few-100 bytes and thus may fail.
Not all memory allocations are checked to be successful, and thus may cause a crash if the program just assumes the allocation was successful.

Can you give a list of plugins running on nodes that crash when MQTT is enabled?
It would be great if you also could show the timing stats of such nodes (must be enabled on the tools-Advanced page) as blocking code can also lead to WiFi disconnects. 
Strictly speaking everything > 10 msec may affect WiFi, but I mainly focus on blocking code > 100 msec.

ghtester, 2022-01-21:
Thanks for the detailed comment. Yes I know other plugins may also play a role on this issue. But I am not using the LCD plugin on all nodes and there are other ESP nodes also crashing in relation to MQTT controller without LCD. So more likely it could be a binding with the IR receive plugin which I am using on most nodes.
Regarding to WiFi - I don't see it should be now the major issue. As outlined earlier, I am observing crashes also when the WiFi connection is OK but the connection to MQTT broker is dropping.  But I'll try to look at timing stats later.
So far I made testing with 3 different MQTT brokers, 2 remote and 1 local with the very similar or the same results (if I am invoking losing connection to local MQTT broker)..
Tested on different ESP firmware releases, all of them custom Vagrant compilation, also the same results.
On 'production' nodes the data to MQTT controller are sent every 60 s so the crash does not happen very often (and it looks there's a better stability on recent firmware builds) but as soon as changed to 1 s in my test environment, the crash issue is reproducible more easy.
Some other nodes:
[image]
[image]

TD-er, 2022-01-21:
There are 2 things you must keep in mind when drawing conclusions.
- Actual WiFi connected state may be different from what the SDK reports.
- MQTT disconnects are a much more reliable indicator for actual disconnected state of your network

Maybe I should add a check in the code that as soon as the MQTT client reports a disconnect, we may retry to connect only once or twice, or else I should disconnect WiFi and retry to connect to the access point.
This has one important drawback. If for whatever reason the MQTT broker is actually not reachable, then all nodes will be unreachable too as they constantly try to reconnect to WiFi.

Another thing I can do, is to look at the MQTT client connected state for the MQTT controllers instead of the "Network connected" state.

There is at least one thing that comes to mind, which is related to how the MQTT client reconnects.
I have seen that sometimes right after a successful connect it may take a long time before you may successfully send out IP traffic.
For example sending a NTP packet to get the time, may take sometimes several tens of seconds to get a reply.
For this I already added a delay between the moment a connection to an AP is successful and report that we're connected to the rest of the code.
By delaying the first packet a few 100 msec, you can actually speed up the entire process as that packet then can be sent immediately.
But maybe this is sometimes not long enough and I have no idea what may be the cause for this and how to check for it.
Well I do have lots of ideas, but none proven to be correct about this ;)

If you can show some timing stats of nodes experiencing these issues, that would be great.
Also, can you try to log (and chart) the available free memory of those nodes for a longer time, please do.

ghtester, 2022-01-21:
> Maybe I should add a check in the code that as soon as the MQTT client reports a disconnect, we may retry to connect only once or twice, or else I should disconnect WiFi and retry to connect to the access point.

I agree this should be changed. But I would NOT try to WiFi reconnect when just the MQTT disconnect is reported by client.

> This has one important drawback. If for whatever reason the MQTT broker is actually not reachable, then all nodes will be unreachable too as they constantly try to reconnect to WiFi.

Exactly, I was observing this several times. An example - 2 nodes suddenly, at the same time, started WiFi reconnecting despite they had a stable WiFi connection until that moment. One node even rebooted in the end... And the reason was the MQTT broker upgrade / patching so it was unavailable for some time. 
I am sure MQTT connection should NOT have a direct binding to WiFi connection status, both these services should be kept 'independent'.

> Another thing I can do, is to look at the MQTT client connected state for the MQTT controllers instead of the "Network connected" state.

That would be nice and it should help a lot I think.

> By delaying the first packet a few 100 msec, you can actually speed up the entire process as that packet then can be sent immediately.
> But maybe this is sometimes not long enough and I have no idea what may be the cause for this and how to check for it.

I suppose the reason is in SDK. Maybe instead of a fixed delay there could be something like a 'ping' to gateway test. But I know it would need a huge part of code for just a simple thing...

> If you can show some timing stats of nodes experiencing these issues, that would be great.

Currently not as I am not right there. To be updated...

> Also, can you try to log (and chart) the available free memory of those nodes for a longer time, please do.

DTTO... Also I'm not sure how to easily make a free memory chart, so far I was just using Syslog.

TD-er, 2022-01-21:
You can use a sysinfo task for it and log your free memory to whatever controller (even MQTT), so we can see a trend.
Of course we will not see memory usage when the system crashes, but at least it is something.

By the way, sending data to a syslog server is also reported by some users as cause for a crash, just like the correlation of MQTT + crashing.

TD-er, 2022-04-01:
Should be fixed by this PR: https://github.com/letscontrolit/ESPEasy/pull/3987

Close for now, please re-open if it isn't properly fixed.

closed 2022-04-01 by TD-er