letscontrolit/ESPEasy #3921: Syntax: [task#value#transformation#justification] problem

ilexpl01, 2022-01-21:
Hi.

Git Build: mega-20211224_f162ebf

Dummy device defined:

[image]

Values are set with boot of the ESP device:

_on System#Boot do
   timerSet,1,1
   timerSet,2,1
   TaskValueSet DayStartTime,HH,6
   TaskValueSet DayStartTime,MM,30
   TaskValueSet DayStopTime,HH,23
   TaskValueSet DayStopTime,MM,59
   TaskValueSet Temperatura,Start,40
   TaskValueSet Temperatura,Stop,42
endon_

I want to compare DayStartTime#HH combined with DayStartTime#MM to %systm_hm%

While DayStartTime#HH is a single digit I need to transform it to two characters string by adding "0".

To do that I go for a formula:  [DayStartTime#HH#D2]

For testing purposes I transfer the output text to my Domoticz over MQTT:

  Publish domoticz/in,'{"command":"udevice","idx":350,"svalue":"Test A [DayStartTime#HH#D2]"}'
   Publish domoticz/in,'{"command":"udevice","idx":350,"svalue":"Test B [DayStartTime#HH#D2.0]"}'
   Publish domoticz/in,'{"command":"udevice","idx":350,"svalue":"Test C [DayStartTime#HH#D2.1]"}'

The results are as follows:

[image]

What the hell do I do wrong ? :)

TD-er, 2022-01-21:
Just for reference the required links:
- [RTD - TaskValueSet](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=TaskValueSet#taskvalueset)  (needs some update in formatting)
- [Formatting referred values](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=formatting#formatting-refered-values)

How do you have configured the dummy task (nr of decimals) ?

ilexpl01, 2022-01-21:
I've seen that guideline.

_Floating point transformations:

_Dx: Minimal ‘x’ digits zero filled in front of the decimal point, no decimal digits. Same as Dx.0___

[image]

ilexpl01, 2022-01-21:
Test with logentry (same result):
10625: EVENT: Rules#Timer=2,1
10691: ACT  : Publish domoticz/in,'{'command':'udevice','idx':350,'svalue':'Test A  6'}'
10783: ACT  : Publish domoticz/in,'{'command':'udevice','idx':350,'svalue':'Test B  6'}'
10810: ACT  : Publish domoticz/in,'{'command':'udevice','idx':350,'svalue':'Test C 06.0'}'
10834: ACT  : LogEntry,'Test A:  6'
10839: Test A:  6
10851: ACT  : LogEntry,'Test B:  6'
10856: Test B:  6
10871: ACT  : LogEntry,'Test C: 06.0'
10876: Test C: 06.0

TD-er, 2022-01-21:
Those links were not just for you, but I know everyone reading this thread will look up the same links as I just did to check for myself ;)

But I think you have found some bug.
No idea yet when or where this may have been introduced.
Right now I have no idea where in the code this leading zero may be lost, as it seems OK what you are doing.

ilexpl01, 2022-01-21:
I am proud of myself to find the 
[image]
I hope it will be killed/solved soon :)

tonhuisman, 2022-01-21:
I have this test-rule from a previous 'exercise'
```
on test2 do
  let,1,1
  logentry,"value: [var#1#D4]"
  let,1,10
  logentry,"value: [var#1#D4]"
  let,1,100
  logentry,"value: [var#1#D4]"
  let,1,-10
  logentry,"value: [var#1#D4]"
endon
```
This is the result:
```
1011956 : Info   : EVENT: test2
1012021 : Info   : ACT  : let,1,1
1012028 : Info   : ACT  : logentry,"value: 00 1"
1012030 : Info   : value: 00 1
1012032 : Info   : ACT  : let,1,10
1012039 : Info   : ACT  : logentry,"value: 0010"
1012041 : Info   : value: 0010
1012043 : Info   : ACT  : let,1,100
1012049 : Info   : ACT  : logentry,"value: 0100"
1012051 : Info   : value: 0100
1012054 : Info   : ACT  : let,1,-10
1012060 : Info   : ACT  : logentry,"value: 0-10"
1012062 : Info   : value: 0-10
```
So it looks like an off by 1 error 🤔

tonhuisman, 2022-01-21:
And the bus seems to be in the Arduino framework, as `String res(value, decimals);` adds an unexpected leading space if the length of the result is 1 and decimals is 0. (It works as intended if decimals > 0).
Smells like a framework bug, but can easily be remedied by adding a simple `res.trim();` after creating the string.

TD-er, 2022-01-21:
We actually have (or had???) a `toString` function just for this

PR #3923 (2022-01-21): [Formatting] Fix issue with unexpected leading space
Resolves #3921
 
An unexpected leading space seems to be inserted by the Arduino `String(value, decimals)` constructor if the result is length 1 and decimals is 0. Trimming the result solves that.
closed 2022-01-21 by TD-er