letscontrolit/ESPEasy #3922: Display on 2 OLEDs with a generic HTTP request

jerome6994, 2022-01-21:
Good evening
I'm trying to display via a generic HTTP request on 2 OLED screens or even 3 if possible
I put a TCA9548A card.
1 - Does it work?
2 - how to write the http request?

I'm a little lost on Github, I'm sorry if the question has been asked before

Thank you for your help
[image]

tonhuisman, 2022-01-21:
You can address separate tasks by prefixing the command like `[<taskname>].<command>` where <taskname> is the name you defined for the task, and <command> the command you want to be executed there. The `[` `]` are optional., the dot is required if the taskname prefix is used.

TD-er, 2022-01-21:
And to be complete, here the link to the documentation about this:
[Command a specific task for multiple instances of a plugin](https://espeasy.readthedocs.io/en/latest/Reference/Command.html?highlight=instances#command-a-specific-task-for-multiple-instances-of-a-plugin)

tonhuisman, 2022-01-21:
It may be a bit challenging for an ESP8266, memory wise, so have more than 2 of these OLedFramed plugins active at the same time, but it is surely worth trying 😉 if it works, it works 👍

jerome6994, 2022-01-21:
Thank you it works very well I missed this in the doc

So yes I read that 2 screens could pass but 3 complicated
I script on it every minute for testing but after that I'll script every 5 minutes I think.

For the moment it works but at times I have a screen that picks up to see the 3 at the same time and just 1 screen returns to the following script.

What should be done with it to have the 3 screens together?

TD-er, 2022-01-22:
The 1 line config is the largest font we included.

Since the limit of 2 of these displays is mainly a memory issues, you could consider switching to an ESP32.

jerome6994, 2022-01-22:
great thank you so much for the info. 

For the moment the 3 screens are going very well.  

the only negative point is the character size that I would have liked bigger to display a full screen temperature on 2 OLEDs and on the 3rd the normal information.

anyway great job it's great

tonhuisman, 2023-09-01:
Issue is solved, and can be closed.

closed 2023-09-01 by TD-er