letscontrolit/ESPEasy #3926: [Feature Request] Force OTA at the cost of wiping filesystem

blazej222, 2022-01-22:
I have a couple of ESP8285 1M flash devices which act as intelligent power outlet switch & power consumption measurement. 
I compiled a custom firmware with `#define PLUGIN_BUILD_MINIMAL_OTA`  and a couple of plugins that are not included in minimal/normal build. Then I packed it into .gz archive so I could perform OTA on old, original firmware.

 Everything was fine until I wanted to include one more plugin, and I noticed EasyESP tells me OTA is not possible to any firmware, even the smallest 200KB image (`Warning: OTA not possible`). 
Disassembling those devices and soldering Serial pins is rather hard, because the housing is glued together and UART pins are not easily accessible (whole process requires a lot of time)

My idea is to add feature that allows to perform OTA at the cost of wiping filesystem and losing all settings in case there's not enough space on the device to save settings. Currently, it won't let you upgrade because it would have to overwrite filesystem with settings.

Would be a great feature to deal with devices that are generally hard to flash via Serial, and you could always go back to original firmware without disassembling the device.

blazej222, 2022-02-11:
@TD-er Is there a code of 2step uploader somewhere? I was able to properly implement this feature and it worked when flashing `Easyesp modified by me->My custom firmware->Easyesp again`, but when I do `Easyesp modified by me->2step uploader->Easyesp again` 2step uploader crashes when uploading easyesp firmware with `Magic byte is wrong, not 0xE9`

TD-er, 2022-02-11:
Yep, got it somewhere, have to look.
What it does is only start a webserver for the OTA and implements support for SPIFFS and the settings file to know how to connect to WiFi.
It was compiled on core 2.3.0 (or maybe even older?) to make it as small as possible.
Not sure if you can compile it today to be that small.

TD-er, 2022-02-11:
@soif created EspBuddy and it also hosts the source of the [ESPEasyUploaderMega.OTA.ino](https://github.com/soif/EspBuddy/blob/master/sketches/ESPEasyUploaderMega.OTA/ESPEasyUploaderMega.OTA.ino)
You should take a look at his project anyway.

soif, 2022-02-11:
You can find the minimal EspEasy "2step" uploader, from my [espbuddy](https://github.com/soif/EspBuddy/) repo, at:
https://github.com/soif/EspBuddy/tree/master/firmwares

BTW using [espbuddy](https://github.com/soif/EspBuddy/)  to automatically make the 2step upload job, should be painless... Give it a try...

blazej222, 2022-02-12:
So far I came up with #3949 and it works. Tested it on 1M ESP 8285, because I don't have esp8266 unit with 1M flash (only a couple with 4M flash). 

Well, we gotta start somewhere - this change doesn't really wipe settings and make it possible to upload bigger images, but I will try doing that when I have more time :D 

I also noticed that on ESP8285 updating anything through 2Step uploader doesn't work, even without my changes. Neither .bin image or .gz compressed archive worked, and the whole process always crashed with `"Magic byte is wrong"` or `Fatal exception`, basically forcing me to flash via Serial. Don't remeber problems like that on ESP8266

I need to further investigate this issue, and if I confirm it on multiple ESP8285 devices will open an issue on github. Maybe I'll borrow ESP8266 1M unit from a friend of mine and test it.

tonhuisman, 2022-02-12:
> I don't have esp8266 unit with 1M flash (only a couple with 4M flash).

You can install a 1M binary on a 4M unit, and it should behave like a 1M unit, even though, technically, the extra flash is available, it shouldn't (be able) use it.

TD-er, 2022-02-12:
The ESP8285 does use less GPIO pins connected to the flash.
So you probably need to compile using dout as flash mode or else things will go wrong.

tonhuisman, 2023-09-01:
The option has been added and is validated to be working, so this issue can be closed.

closed 2023-09-01 by TD-er