letscontrolit/ESPEasy #3930: IR Receive Task Settings leads to freeze or crash

ghtester, 2022-01-25:
An issue observed on some ESP Easy nodes after upgrade (from earlier builds about 2 months old) to latest Custom build compiled 20220122.

When I try to edit  **Communication - IR Receive (TSOP4838)** plugin's Task Settings from Devices web page, on some nodes (with different configurations / plugins used) these issues happen:
- task configuration page is rendering slowly
- task configuration page stops rendering and the ESP node hangs
- task configuration page stops rendering 

The same or very similar behaviour with access from Chrome and Edge web browsers.

[image]

tonhuisman, 2022-01-25:
Can you add a Sysinfo plugin task, and configure the Free Heap, Load and stack memory?

The IR plugin uses a lot of memory/resources on its own, possibly there is an IR transmitter, in view of your receiver, that overloads the unit?

Does it improve if you disable the DS18b20 sensor?

ghtester, 2022-01-25:
[image]
The IR plugin on this node has no sensor installed so there's no IR input overloading the node.
Unfortunately IR receive becomed a bit too complex / memory hungry from my point of view, including the configuration page - I am using the IR receive on most nodes but never needed a command mappings (added couple months ago) which could be useful for some users but it's not sufficient for me anyway so I am using rules (which are also easy to copy / paste between nodes in opposite to IR receive command mappings).

Disabling DS18b20 helped, now the confoguration page could be rendered completely (but slowly) without crash /hang.

tonhuisman, 2022-01-25:
Hm, your `freeheap` result is rather low. I assume it will improve quite a bit if you disable the unused IR, to avoid possible spuriously detected fake-input for that plugin, that might also help.

ghtester, 2022-01-25:
OK but it's not a solution for me. On this node it was installed just for a stability test, on most nodes I need it.

TD-er, 2022-01-25:
There will be a fix soon, as the PR for SDK3.0 is almost ready to be merged and "beta" builds on that PR will have quite a bit more memory.

TD-er, 2022-01-25:
And of course, if those new items of the plugin are causing issues, they should be made optional in the build wrapped in #ifdef checks.

ghtester, 2022-01-25:
Thank you for your feedback and hope. It would be great to see that kind of progress. The low available memory is a most often reason for crashing.
Thanks to effort of all develpers & contributors, we see a great progress in ESP Easy functions / plugins / devices supported but the stability is still an issue unfortunately, at least in my use cases.

TD-er, 2022-01-25:
I will have a look at that plugin code.
If it generates a huge list of strings, that could for sure lead to memory issues.
Not sure if that can be fixed in ESPEasy, or should be fixed in the library, but the maintainer of that library is always very willing to cooporate .

TD-er, 2022-01-25:
Can you show me a screenshot of those 2 lists?
Just looked at the code and it seems indeed as a struct which may need some extra attention.
How many of those items are there?

tonhuisman, 2022-01-25:
> How many of those items are there?

I'll look into fixing that using a html `datalist` if possible, I've been using that for another, yet unpublished, feature too.

ghtester, 2022-01-25:
[image]
Both lists are the same, do multiple the window listed by about 5 times...

tonhuisman, 2022-01-25:
These lists are dropped to the browser as an `<options>` list for each combo, AFAICS, not kept in memory for that long. And most likely the browser can handle that without issue.

ghtester, 2022-01-25:
The browser can handle a lot of data for sure but the question is which amount of data is capable the ESP node to serve...
The issue is easily reproducible anyway. The MQTT controller, the IR Receive and some other plugin(s) configured and that's it...

tonhuisman, 2022-01-25:
IR Receive is known to be quite 'heavy', that's the main reason the IR related stuff is in a separate build, with nearly no other plugins/controllers, to avoid these kind of issues. But the IR library list of supported devices keeps growing...
You could switch to ESP32 units, there it is much less of an issue, as more memory (320 kB on regular ESP32 WROOM/WROVER) is available 😃

TD-er, 2022-01-25:
One thing I noticed, is that it creating a completely new P016_data_struct and loading the current settings into that one.
That's fine if the task wasn't active/enabled, but as soon as it is, this may cause issues.

@tonhuisman It is somewhat clear why you create one for loading (but that could be done more efficient) the page?
But it is no longer needed to keep the existing one on saving, as right after saving the task is restarted anyway.

During `PLUGIN_LOAD` you don't need the entire data structure, as you also load it per map entry from the file system.
So instead of loading it all into a giant structure, you can also load one item at a time, create the HTML and load the next.

ghtester, 2022-01-25:
> You could switch to ESP32 units, there it is much less of an issue, as more memory (320 kB on regular ESP32 WROOM/WROVER) is available 😃

NOPE, I won't replace my ESP8266 nodes for sure. :-) Especially when I am sure just a slight SW improvement could fix the issue (and I believe the IR receive code itself is OK, just the things around, like the config page, are not optimal). If that assumption is true (there are non-optimized code parts),  couple months later we'll be in the same situation with ESP32... ;-)
 Unfortunately I am not able to rewrite the code by myself so I hope some of the great programmers here (= you) could perhaps  help... ;-)

TD-er, 2022-01-25:
> Unfortunately I am not able to rewrite the code by myself so I hope some of the great programmers here (= you) could perhaps help... ;-)

Debugging and fixing is removing of bugs...
So what do you think is what is done during programming?

ghtester, 2022-01-25:
I really don't know how programmers work now (and it's sometimes hard to understand when I see for instance the quality of many even professional apps for MS Windows) but  I think it very depends on concrete person / team. ;-)
 In the deep past the programmer first thought about it, then wrote the program code and the debugging/optimizing was the third stage. :-)

But to be serious - I know it's not easy to write even one small application. Not to mention a project like ESP Easy. It's a really great work, what you are doing and I am sure many happy users are appreciating it a lot.

TD-er, 2022-01-25:
I will have a look at it and also discuss with @tonhuisman to what would be the best way to fix it, so we have fresh new bugs then ;)

TD-er, 2022-01-26:
See: https://github.com/letscontrolit/ESPEasy/pull/3932
Just wrote down my thoughts while watching Dexter, so have to take a good look at it tomorrow, with a fresh mind.

ghtester, 2022-01-26:
A more then expected. ;-) Thanks a lot for your effort!

PR #3932 (2022-01-27): [P016] Reduce memory usage during load/save (#3930)
Just a quick write down of my idea to reduce memory usage, while watching TV...
Somehow it will not yet compile.

Todo:
- [x] Make it compile
- [ ] Test

Fixes: #3930
closed 2022-01-27 by TD-er