letscontrolit/ESPEasy #3931: No access after setting a password on ESP32

xxxturboxxx, 2022-01-25:
I'm using prelnty of ESP8266 board and now I tryed some ESP32 as well. I've flashed the latest release 20211224 on each module and everything worked fine til I set up a password for each board (same PW for all boards). 

It works well on all ESP8266 boards but none of the ESP32 are accessable anymore.

Tryed admin and Admin + the selected password. 

Is there a difference between ESP8266 and ESP32?

Thanks for your support!

tonhuisman, 2022-01-25:
Please have a look at issue #3887 where exactly the same issue occurred. This is a bug in the ESP32 Arduino framework, that has been fixed already but not in use by ESPEasy yet (but soon will be), and has been resolved some time ago for ESP8266.

Quick fix: Use Internet Explorer (or another very old web browser) to login to your ESP32.

xxxturboxxx, 2022-01-25:
Thanks for the quick support and sorry for mentioning the same issue twice. I've searched for this issue but could not find anything. Need to search better :-)

TD-er, 2022-01-25:
Close this one as it is a duplicate.

closed 2022-01-25 by TD-er