letscontrolit/ESPEasy #3936: OLED SSD1306: Lines per frame limited to 3 since Version "Release mega-20201227"

wisgit, 2022-01-30:
Here is my sysinfo-copy-to-clipboard:
```
Build:⋄
20111  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄

Plugin Count:⋄
46 [Normal]
Build Time:⋄
Nov 30 2020 00:38:42
Binary Filename:⋄
ESP_Easy_mega_20201130_normal_ESP8266_4M1M
Build Platform:⋄
Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20201130_f4f960c
```

### Summarize of the problem/feature request

Up to firmware "ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin" more than 3 line per framewere  possible for the OLED SSD1306. With firmware "ESP_Easy_mega_20201227_normal_ESP8266_4M1M.bin" and later only <= 3 lines per frame are possible.
(I know that >=4 lines per frame are hard to read, but it was possible at least)

tonhuisman, 2022-01-30:
Have you tried the [latest release](https://github.com/letscontrolit/ESPEasy/releases)?

What is the Size (resolution) and Rotation of the display you are using?

wisgit, 2022-01-30:
The size is 128x64, rotation is "Rotated".
I actually upgraded from an older version (sometime in 2020) to version "ESP_Easy_mega_20211224_normal_ESP8266_4M1M.bin", when I noticed the issue. Then I stepwise downgraded to find out at which version the issue first occurred.

TD-er, 2022-01-30:
Make a backup of your settings and try to recreate the settings on that task on a newer build.
As this is by far one of the more popular plugins, I am quite confident it may be an issue in converting the settings and not on the operation of the plugin as others would have mentioned it before.

It may also help to open the settings page of that task and save it again (running a newer build)

wisgit, 2022-01-31:
I saved settings, updated firmware, reloaded the settings, still the same, reloaded again, rebooted, still the same.
Then, under new firmware:
saved settings, reloaded settings, rebooted: still the same
Is there any information which I could/should provide?
(For me the matter is not urgent, I can well live with the older firmware)

Finally: A great thanks to all of you who do this work for a community!

TD-er, 2022-01-31:
What I meant was, you can backup the config.dat so you can restore it if you decide to revert to the older build.

Now with the newer build active, you can delete the display task, then add it again and setup the fields you need for the display task. (so no need to change the rest)

TD-er, 2022-01-31:
OK, I just tested it on a unit running the latest code and indeed it only shows 3 lines, not 4. Even whet set to 4.

@tonhuisman Something for you to look at?

tonhuisman, 2022-01-31:
I have replaced a font that was never selected, by a font small enough to re-enable the 4-line frames.

@wisgit Are you available for testing? You can download from the Github actions run (once it's ready) in the [pull request](https://github.com/letscontrolit/ESPEasy/actions/runs/1775154207) (valid until the next commit on that PR 😉).

wisgit, 2022-02-01:
@tonhuisman It's working!
Thanks a lot!

tonhuisman, 2022-02-01:
> bonus question: How would I set the 'lines per frame' from {some external source}

I'm afraid that's not possible, atm. The plugin doesn't support the `config` command, nor handles a command to set that.
I could add that, even though it is outside the context of a bugfix 🤔, I'll await the verdict of @TD-er to add this now, or postpone until later. (Small addition, but still increases the code)

TD-er, 2022-02-01:
It is not a big increment of the code and I think we've run into a lot of other size issues anyway so we need to change the build definitions anyway as it will not fit anymore  (see the large number of failed builds lately, all due to size)

tonhuisman, 2022-02-01:
It needs a bit more code than initially anticipated, on ESP8266 it adds 368 bytes to the .bin size (can be disabled using a compile-time define if needed for size-challenged builds). Also can generate an event, like other events are optionally generated.
Updated the documentation.

Downloads can be obtained [here](https://github.com/letscontrolit/ESPEasy/actions/runs/1780617375) after the Github action is completed.

wisgit, 2022-02-05:
I didn't expect that my remarks would cause so much work, sorry for that. Especially since my idea to have 4 lines was rather spontaneously (I wanted to display wind & gust speeds during the storm we had lately). 

Now, after reverting to 3 lines as I had before, I noticed that with Ton's update the small font he had activated in order to accomodate 4 lines is also used for the 3-line mode. That's a drawback, IMO, as the "ESP_Easy_mega_20201130" version had a larger and better to read font for 3 lines. I've downgraded to "ESP_Easy_mega_20201130" therefore.
My suggestion is to leave everything as it was before I stirred that up, and maybe enter a remark in the docs that >3 lines are not supported do to bad readability  ...   ;-)

Sorry again, 
Franz

tonhuisman, 2022-02-05:
> I didn't expect that my remarks would cause so much work, sorry for that. Especially since my idea to have 4 lines was rather spontaneously (I wanted to display wind & gust speeds during the storm we had lately).

The amount of work is not really an issue, this is a hobby for me, programming 😄 

> Now, after reverting to 3 lines as I had before, I noticed that with Ton's update the small font he had activated in order to accomodate 4 lines is also used for the 3-line mode. That's a drawback, [...]

The actual description for that is 'bug', so that has to be fixed. I hadn't noticed, or just didn't look close enough, but I should have.
Sorry for that.

> My suggestion is to leave everything as it was before I stirred that up, and maybe enter a remark in the docs that >3 lines are not supported do to bad readability ... ;-)

Well, it hadn't been reported before, but still it should be fixed, as a) it has worked that way before, and we really want to be backward compatible, and b) we hate bugs 😉

tonhuisman, 2022-02-05:
> The actual description for that is 'bug', so that has to be fixed. I hadn't noticed, or just didn't look close enough, but I should have. Sorry for that.

Been doing some debugging, but wat actually is happening is that in the older build, the lines where drawn somewhat (1px) closer together, as for 4 lines to fit I had to enlarge the content space. For as far as I can see, a different font is used for the 4-line frames compared to the 3-lines frames.
So, no bug, according to my investigation.

logging (added for debugging):
```
63517 : Info   : HTTP: oledframedcmd,linecount,4
63522 : Info   : P036 CalculateFontSettings lines: 4
63523 : Info   : CalculateFontSettings prLines: 4, max: 10, i: 4, h: 10, f: 3, font: 3
...
116457 : Info   : HTTP: oledframedcmd,linecount,3
116462 : Info   : P036 CalculateFontSettings lines: 3
116463 : Info   : CalculateFontSettings prLines: 3, max: 14, i: 4, h: 10, f: 3, i: 3, h: 13, f: 2, i: 2, h: 19, font: 2
```
(prLines is the number of lines requested, max is the largest fontheight allowed, i is the current index, h is the height of the font being tested, f is the font tested. From i on the values are repeated for the next loop) 
Fonts are internally numbered 0..3, where 3 is the smallest font, and 0 the largest.

wisgit, 2022-02-05:
Here is what I observe: 
2 screenshots  with 3 and 4 lines, resp., taken with firmware "SP_Easy_mega_20201130_normal_ESP8266_4M1M.bin":
[image]
[image]

And 2 screenshots  with 3 and 4 lines, resp., taken with firmware "ESP_Easy_mega_20220131_normal_ESP8266_4M1M":
[image]
[image]

TD-er, 2022-02-05:
Wow, that's an ugly rendering of the font.

tonhuisman, 2022-02-05:
Thanks for the great de example photos.

> ugly rendering of the font.

That is probably why that font was not in the code before, and I'd probably should get rid of it again 😉 
It also shows that the 2020 build used a different, larger font. It was probably changed to make room for the indicator, showing dots for the pages and a circle for the active page. But that is only activated on multiple pages. And also, some extra space between the lines was added, that is probably why 4 lines wouldn't fit anymore.

I'm now trying to fit in the nicer fonts, by leaving out the extra space between the lines, but the code is a bit hard to grasp in some places.

tonhuisman, 2022-02-12:
@wisgit I've changed the font used for the 4-lines display (again), this time to a non-proportional font but with better readability, can you re-test, please, to see if it still works as expected and intended?

tonhuisman, 2022-02-13:
> They all produce the same font as in picture no. 4

For the 4-line display, with the Indicator enabled, I have to use an extra small font to be able to fit the lines in the space available. If you disable the indicator in the settings, the same font as used for the previously shown 3-lines display should fit for the 4-lines, giving the better readable display.

NB: If you have only 4 lines of information (like you seem to have), then the Indicator is not shown, but its space is still reserved. I can't auto-disable the indicator, as when adding more lines of information it would then again re-enable the indicator, resulting in the smaller font being used, hence I decided to make that a separate setting.

wisgit, 2022-02-13:
Thanks for all the effort!
But what I don't understand:
Why was it possible up to version "ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin" to have, in addition to the header line, 4 lines with the header-line font, plus indicators, as shown in the figure below (sorry for the bad quality):
[image]

PR #3938 (2022-04-04): [P036] Bugfix: Re-enable 4-line frames, Feature: Left-align content
Resolves #3936 
Resolves #3943 

- By tweaking the height of the footer and the font metrics, the 4-line frames can now be displayed again.
- Bonus 1: Add sub-command `linecount` for changing the number of lines per frame. Activates on next Interval. (Disabled on TEST and MINIMAL builds)
- Bonus 2: Add setting to optionally hide the Indicator (dots showing page/active page in footer), so this space can be used for content, often using a larger font
- Bonus 3: Add Left-align option for content, including `leftalign` sub-command [... cut]
closed 2022-04-04 by TD-er