letscontrolit/ESPEasy #3939: No Longer Compil when FEATURE_ARDUINO_OTA is set

soif, 2022-02-01:
The Latest 'normal_ESP8266_4M1M'  no longer succeeds to compil, when the FEATURE_ARDUINO_OTA is set.
It returns : " **'.text1' will not fit in region iram1_0_seg** "

_(FYI, It was still working fine in the Git Version as of 23 dec 2021)_

## How to reproduce

```
export PLATFORMIO_BUILD_FLAGS='-DFEATURE_ARDUINO_OTA'
/usr/local/bin/pio run -e normal_ESP8266_4M1M 
```

## Espeasy / platformIO versions
```
mbp1-eth:ESPEasy moi$ git show
commit fb03dca5e31c662272d4fc46740fffbe36c07547 (HEAD -> mega, origin/mega, GH/mega)
Merge: cbac54370 26a91039b
Author: TD-er <gijs.noorlander@gmail.com>
Date:   Sat Jan 29 15:40:20 2022 +0100

mbp1-eth:ESPEasy moi$ /usr/local/bin/pio --version
PlatformIO Core, version 5.2.4
```

## Compilation without flag (SUCCESS):

```
mbp1-eth:ESPEasy moi$  export PLATFORMIO_BUILD_FLAGS=''
mbp1-eth:ESPEasy moi$  /usr/local/bin/pio run -e normal_ESP8266_4M1M 
Processing normal_ESP8266_4M1M (platform: espressif8266@2.6.3; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
 Default PIO configuration check 
 CPPDEFINES:   ['NDEBUG', 'VTABLES_IN_FLASH', 'PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH', ['PUYA_SUPPORT', '1'], 'CORE_POST_2_5_0', 'BEARSSL_SSL_BASIC', 'CORE_POST_2_6_0', ['PSTR_ALIGN', '1'], 'PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', ['MQTT_MAX_PACKET_SIZE', '1024'], ['HTTPCLIENT_1_1_COMPATIBLE', '0']]
 ------------------------------- 
 
... (log removed).......

 ------------------------------- 
 Compile time defines 
 File System:      SPIFFS
 PROGNAME:         ESP_Easy_mega_20220201_normal_ESP8266_4M1M
 BUILD_PLATFORM:   macOS-10.13.6-x86_64-i386-64bit
 GIT_HEAD:         pygit2_not_installed
 ------------------------------- 
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.3) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.7.4+sha.2843a5a 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)

... (log removed).......

Linking .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf
Building .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin
Retrieving maximum program size .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf
Checking size .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  56.7% (used 46476 bytes from 81920 bytes)
Flash: [========= ]  90.4% (used 943888 bytes from 1044464 bytes)
Creating BIN file ".pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin" using "/Users/moi/.platformio/packages/framework-arduinoespressif8266@src-08ac9a8f0276d3842d078aba1dc9d0b3/bootloaders/eboot/eboot.elf" and ".pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf"
bin_gzip([".pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin"], [".pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf"])
 gzip:   .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin
bin_elf_copy([".pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin"], [".pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf"])
 copy to:   build_output/debug/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf
 copy to:   build_output/bin/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin
 JSON to:   build_output/json/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.manifest.json
 copy to:   build_output/bin/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.bin.gz
 copy to:   build_output/debug/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.env.txt
 Timestamp: 2022-02-01 11:52:41.033112
============================================ [SUCCESS] Took 350.87 seconds ============================================

Environment          Status    Duration
-------------------  --------  ------------
normal_ESP8266_4M1M  SUCCESS   00:05:50.868
============================================= 1 succeeded in 00:05:50.868 =============================================

```

## Compilation with FEATURE_ARDUINO_OTA flag (FAILED):

```
mbp1-eth:ESPEasy moi$ export PLATFORMIO_BUILD_FLAGS='-DFEATURE_ARDUINO_OTA'
mbp1-eth:ESPEasy moi$  /usr/local/bin/pio run -e normal_ESP8266_4M1M 
Processing normal_ESP8266_4M1M (platform: espressif8266@2.6.3; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
 Default PIO configuration check 
 CPPDEFINES:   ['NDEBUG', 'VTABLES_IN_FLASH', 'PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH', ['PUYA_SUPPORT', '1'], 'CORE_POST_2_5_0', 'BEARSSL_SSL_BASIC', 'CORE_POST_2_6_0', ['PSTR_ALIGN', '1'], 'PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', ['MQTT_MAX_PACKET_SIZE', '1024'], ['HTTPCLIENT_1_1_COMPATIBLE', '0'], 'FEATURE_ARDUINO_OTA']
 ------------------------------- 

... (log removed).......

 ------------------------------- 
 Compile time defines 
 File System:      SPIFFS
 PROGNAME:         ESP_Easy_mega_20220201_normal_ESP8266_4M1M
 BUILD_PLATFORM:   macOS-10.13.6-x86_64-i386-64bit
 GIT_HEAD:         pygit2_not_installed
 ------------------------------- 
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.3) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.7.4+sha.2843a5a 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)

... (log removed).......

Linking .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf
/Users/moi/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [.pio/build/normal_ESP8266_4M1M/ESP_Easy_mega_20220201_normal_ESP8266_4M1M.elf] Error 1
============================================= [FAILED] Took 380.49 seconds =============================================

Environment          Status    Duration
-------------------  --------  ------------
normal_ESP8266_4M1M  FAILED    00:06:20.490
======================================== 1 failed, 0 succeeded in 00:06:20.490 ========================================
```

TD-er, 2022-02-01:
As a matter of fact, I was just looking into something related, as all the builds on Github Actions recently fail because of binary size, where they fit just fine when I build on Windows.
I guess macOS is using nearly the same compiler as is used on Linux.

Can you checkout [this PR](https://github.com/letscontrolit/ESPEasy/pull/3937) and see if it also fails when using that flag?

TD-er, 2022-02-01:
I just checked it myself on a Linux machine and can reproduce your issue.
Also testing against that PR, does not fix it if you want to include Arduino OTA

soif, 2022-02-01:
Thanks for the Feddback. Glad that you've confirmed this bug...

Let me know if you need me to make further tests.

TD-er, 2022-02-01:
Maybe you can redirect the output of the verbose compile run to a file, so I can compare the actually used compiler flags?
I got the impression that GCC on Windows does honor some alignment stuff, or at least handles it different from GCC on Unix/Linux.

soif, 2022-02-01:
Is this what you need ? _(else please advice)_

```
mbp1-eth:ESPEasy moi$  /usr/local/bin/pio run -e normal_ESP8266_4M1M -v 
Processing normal_ESP8266_4M1M (extends: esp8266_4M1M; platform: espressif8266@2.6.3; platform_packages: framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.7.4; build_flags: -DNDEBUG, -mtarget-align, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++, -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH, -DPUYA_SUPPORT=1, -DCORE_POST_2_5_0, -fno-strict-aliasing, -O2, -DBEARSSL_SSL_BASIC, -DCORE_POST_2_6_0, -DPSTR_ALIGN=1, -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703, -Wno-deprecated-declarations, -DMQTT_MAX_PACKET_SIZE=1024 -DHTTPCLIENT_1_1_COMPATIBLE=0, -DFEATURE_ARDUINO_OTA; board: esp12e; board_build.flash_mode: dout; board_upload.maximum_size: 1044464; board_build.ldscript: eagle.flash.4m1m.ld; board_build.f_cpu: 80000000L; build_unflags: -DDEBUG_ESP_PORT; lib_deps: td-er/ESPeasySerial @ 2.0.8, adafruit/Adafruit ILI9341 @ ^1.5.6, Adafruit GFX Library, LOLIN_EPD, Adafruit BusIO, bblanchon/ArduinoJson @ ^6.17.2, VL53L0X @ 1.3.0, SparkFun VL53L1X 4m Laser Distance Sensor @ 1.2.9, td-er/RABurton ESP8266 Mutex @ ^1.0.2, td-er/SparkFun MAX1704x Fuel Gauge Arduino Library @ ^1.0.1, ESP8266HTTPUpdateServer, FrogmoreScd30, Multi Channel Relay Arduino Library, SparkFun ADXL345 Arduino Library, ITG3205; lib_ignore: ESP32_ping, ESP32WebServer, ESP32HTTPUpdateServer, ServoESP32, IRremoteESP8266, HeatpumpIR, TinyWireM, IRremoteESP8266, HeatpumpIR, LittleFS(esp8266), ServoESP32, TinyWireM; monitor_filters: esp8266_exception_decoder; lib_ldf_mode: deep+; lib_archive: False; framework: arduino; upload_speed: 115200; monitor_speed: 115200; targets: ; extra_scripts: pre:tools/pio/pre_default_check.py, tools/pio/gzip-firmware.py, pre:tools/pio/set-ci-defines.py, pre:tools/pio/concat_cpp_files.py, pre:tools/pio/generate-compiletime-defines.py, tools/pio/copy_files.py, post:tools/pio/remove_concat_cpp_files.py; src_filter: +<*> -<.git/> -<.svn/> -<example/> -<examples/> -<test/> -<tests/> -<*/Commands/> -<*/ControllerQueue/> -<*/Globals/> -<*/Helpers/> -<*/PluginStructs/>  -<*/WebServer/>)
------------------------------------------------------------------------------------------------------------------------
 Default PIO configuration check 
 CPPDEFINES:   ['NDEBUG', 'VTABLES_IN_FLASH', 'PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH', ['PUYA_SUPPORT', '1'], 'CORE_POST_2_5_0', 'BEARSSL_SSL_BASIC', 'CORE_POST_2_6_0', ['PSTR_ALIGN', '1'], 'PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', ['MQTT_MAX_PACKET_SIZE', '1024'], ['HTTPCLIENT_1_1_COMPATIBLE', '0'], 'FEATURE_ARDUINO_OTA']
 ------------------------------- 
 Concat ./src/src/Commands/*.cpp to ./src/src/Commands_tmp/__tmpfile.cpp 
 Add:   ./src/src/Commands/Timer.cpp
 Add:   ./src/src/Commands/Controller.cpp
 Add:   ./src/src/Commands/i2c.cpp
 Add:   ./src/src/Commands/Diagnostic.cpp
 Add:   ./src/src/Commands/RTC.cpp
 Add:   ./src/src/Commands/wd.cpp
 Add:   ./src/src/Commands/Tasks.cpp
 Add:   ./src/src/Commands/Common.cpp
 Add:   ./src/src/Commands/System.cpp
 Add:   ./src/src/Commands/Networks.cpp
 Add:   ./src/src/Commands/GPIO.cpp
 Add:   ./src/src/Commands/Time.cpp
 Add:   ./src/src/Commands/SDCARD.cpp
 Add:   ./src/src/Commands/WiFi.cpp
 Add:   ./src/src/Commands/Blynk.cpp
 Add:   ./src/src/Commands/HTTP.cpp
 Add:   ./src/src/Commands/MQTT.cpp
 Add:   ./src/src/Commands/Rules.cpp
 Add:   ./src/src/Commands/Notifications.cpp
 Add:   ./src/src/Commands/Servo.cpp
 Add:   ./src/src/Commands/Settings.cpp
 Add:   ./src/src/Commands/UDP.cpp
 Add:   ./src/src/Commands/InternalCommands.cpp
 Files not changed, will not touch __tmpfile.cpp 
 ------------------------------- 
 Concat ./src/src/ControllerQueue/*.cpp to ./src/src/ControllerQueue_tmp/__tmpfile.cpp 
 Add:   ./src/src/ControllerQueue/queue_element_formatted_uservar.cpp
 Add:   ./src/src/ControllerQueue/C016_queue_element.cpp
 Add:   ./src/src/ControllerQueue/SimpleQueueElement_string_only.cpp
 Add:   ./src/src/ControllerQueue/DelayQueueElements.cpp
 Add:   ./src/src/ControllerQueue/C018_queue_element.cpp
 Add:   ./src/src/ControllerQueue/C011_queue_element.cpp
 Add:   ./src/src/ControllerQueue/queue_element_single_value_base.cpp
 Add:   ./src/src/ControllerQueue/C015_queue_element.cpp
 Add:   ./src/src/ControllerQueue/MQTT_queue_element.cpp
 Files not changed, will not touch __tmpfile.cpp 
 ------------------------------- 
 Concat ./src/src/Globals/*.cpp to ./src/src/Globals_tmp/__tmpfile.cpp 
 Add:   ./src/src/Globals/MainLoopCommand.cpp
 Add:   ./src/src/Globals/Cache.cpp
 Add:   ./src/src/Globals/RuntimeData.cpp
 Add:   ./src/src/Globals/RTC.cpp
 Add:   ./src/src/Globals/Logging.cpp
 Add:   ./src/src/Globals/NPlugins.cpp
 Add:   ./src/src/Globals/Protocol.cpp
 Add:   ./src/src/Globals/ResetFactDefaultPref.cpp
 Add:   ./src/src/Globals/CPlugins.cpp
 Add:   ./src/src/Globals/Plugins.cpp
 Add:   ./src/src/Globals/Plugins_other.cpp
 Add:   ./src/src/Globals/TXBuffer.cpp
 Add:   ./src/src/Globals/ESPEasyWiFiEvent.cpp
 Add:   ./src/src/Globals/CRCValues.cpp
 Add:   ./src/src/Globals/Services.cpp
 Add:   ./src/src/Globals/ExtraTaskSettings.cpp
 Add:   ./src/src/Globals/Statistics.cpp
 Add:   ./src/src/Globals/NetworkState.cpp
 Add:   ./src/src/Globals/EventQueue.cpp
 Add:   ./src/src/Globals/MQTT.cpp
 Add:   ./src/src/Globals/TimeZone.cpp
 Add:   ./src/src/Globals/I2Cdev.cpp
[... 390 lines cut ...]
test_A_ESP32_4M316k_ETH                        IGNORED
[... cut ...]
max_ESP32_16M8M_LittleFS_ETH                   IGNORED
spec_debug_custom_ESP8266_4M1M                 IGNORED
spec_debug_custom_IR_ESP8266_4M1M              IGNORED
spec_debug_beta_custom_ESP8266_4M1M            IGNORED
spec_debug_custom_ESP32_4M316k                 IGNORED
spec_debug_max_ESP32_16M8M                     IGNORED
spec_memanalyze_ESP8266                        IGNORED
======================================== 1 failed, 0 succeeded in 00:02:28.638 ========================================
mbp1-eth:ESPEasy moi$ 
```

uzi18, 2022-02-01:
> Maybe you can redirect the output of the verbose compile run to a file, so I can compare the actually used compiler flags? I got the impression that GCC on Windows does honor some alignment stuff, or at least handles it different from GCC on Unix/Linux.

Should be reported to platformio - behavior should be same

TD-er, 2022-02-01:
> Should be reported to platformio - behavior should be same

Yep, but right now I have no idea where the 'bug' is, so no idea where to report it.

uzi18, 2022-02-02:
@TD-er maybe it is a good start:
```
   -frecord-gcc-switches
       This switch causes the command line that was used to invoke the compiler to
       be recorded into the object file that is being created.  This switch is only
       implemented on some targets and the exact format of the recording is target
       and binary file format dependent, but it usually takes the form of a section
       containing ASCII text.
```

https://stackoverflow.com/questions/12112338/get-the-compiler-options-from-a-compiled-executable

tonhuisman, 2023-09-09:
This seems to be long solved, so can be closed.

closed 2023-09-09 by TD-er