letscontrolit/ESPEasy #3941: no "system#boot" on Build: ESP_Easy_mega_20211105_normal_ESP8285_1M

Sminter4, 2022-02-02:
Hello everyone, actually a simple task.
I flashed a Sonoff S26 to define my own rules.
Sometimes my kids forget to turn off the hay fan. So I defined the following rule for an automatic shutdown after 30 minutes:

```
on System#Boot do
 timerset,1,4
endon 

On Button#State=1 do
  if [Relay#State]=0
    event,on
  else
    event,off
  endif
endon

on on do
 gpio,12,1  //Relay on
 gpio,13,0  //LED on
 timerset,1,1800
endon

on off do
 gpio,12,0
 gpio,13,1
 timerset,1,0
endon

on Rules#Timer=1 do  //auto off
  event,off
endon
```
Unfortunately the trigger "on system#boot do" doesn't work in this build. So that in the event of a power failure and the subsequent power supply, the fan heater switches on by itself and no longer switches off!
Does anyone have a solution?

tonhuisman, 2022-02-02:
- Has it worked in earlier builds?
- Have you tried on the latest release 20211224 ?
- Maybe 4 seconds is a bit too short after boot, and/or the state is also changed elsewhere, _after_ that 4 seconds have expired? You can confirm that by using `loopTimerSet,1,4` so it gets triggered every 4 seconds until the timer is reset by the `off` event.

TD-er, 2022-02-02:
Just my initial thought... can you rename the events?
No idea if `on` as event name may cause issues.

Sminter4, 2022-02-02:
Thank you for your prompt reply.

flashing at sonoff is very cumbersome (only serial, no OTA). So I haven't tried any other build yet.

The shutdown does not work with or without "Send boot state".

Increasing the initial timer (10s) or replacing it with the "event off" is also unsuccessful.

On the log page I see that there is no trigger for `system#boot`.
Each command on its own works correctly though (via the Command page). The timer, the events, etc...

I have now replaced the event `on` with the German `an`. Unfortunately there is no change.

tonhuisman, 2022-02-02:
You can't see the `System#Boot` event in the weblog as it happens very early in the startup of the ESP, before WiFi is connected. And you can't look back in the log from where you started (= connected via WiFi).

Have you tried the `LoopTimerSet,1,4` suggestion I gave?

Sminter4, 2022-02-02:
Yes, I tested the looptimer. LoopTimerSet,1,4 instead of timerset,1,4. Unfortunately this doesn't work either.

With another IP socket (GOSUND) I use the build ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.bin. Here the original (above) code works.

If the solution is only feasible with another build, I have the huge effort to flash all my sonoffs again.

tonhuisman, 2022-02-02:
> If the solution is only feasible with another build, I have the huge effort to flash all my sonoffs again.

Any fix supplied via a new build will have that 'effect'... 🤔 

Another try could be:
Instead of `on System#Boot do` event, you can implement `on WiFi#Connected do` with a guard:
```
on WiFi#Connected do
  if %v123% = 0 // NB: 123 can be any int value > 0, the vars will all be reset to 0 on boot
    TimerSet,1,4
    let,123,1
  endif
endon
```

TD-er, 2022-02-02:
Hmm that makes me curious, to what was apparently broken (and fixed) since that build of November 2021.

tonhuisman, 2022-02-02:
> Hmm that makes me curious, to what was apparently broken (and fixed) since that build of November 2021.

Didn't have time to look at that (yet).

Sminter4, 2022-02-02:
> Instead of `on System#Boot do` event, you can implement `on WiFi#Connected do` with a guard:

Thanks for the great workaround! It works in a small test. :-)

Of course, I haven't been able to test an actual power failure yet. If the WLAN router freezes, the fan heater doesn't switch off and burns my house down. :-) However, many factors would have to come together for this...

TD-er, 2022-02-02:
There should also be a `System#Wake` event.

TD-er, 2022-02-02:
And you can also check on events received by other tasks (or create one, like a sysinfo task) to see if a var is set.
Just as an extra precaution.

Sminter4, 2022-02-02:
> System#Wake

Just tested. That does not work either.

It's correct. I could still query the uptime and generally switch off uptime after one minute.

Sminter4, 2022-02-19:
I've also tried the "ESP_Easy_mega_**20211224**_normal_ESP8285_1M.bin" build. 
As with the previous version, `system#boot` does not work here either. Did you already find the error for this?

tonhuisman, 2022-02-19:
Can you try to turn off the Rules option on the Tools/Advanced page, save that setting, then turn it back on, and again save it?
Then please re-test.
As the only condition for that event to be generated is this settings value, looks like this:
```
  if (Settings.UseRules)
  {
    String event = F("System#Boot");
    rulesProcessing(event); // TD-er: Process events in the setup() now.
```
But that same check is in at least 50 other places in the code, so...

tonhuisman, 2022-02-19:
Another debugging attempt:
- Add a Dummy Device, you could name that `test`, and enable it
  - If you don't have a Dummy device available in your build, then you can use a Switch device, but you will have to enable `Allow TaskValueSet on all plugins` on the Tools/Advanced page, and replace `Dummy` with `State` in the TaskValueSet command, below.
- In the `on System#Boot do` event add `TaskValueSet,test,Dummy,1
- Reboot the device
- Check on the Devices page if the added `test` task shows value 1 (as it should be set by the Boot event)

NB: Just tested this myself on an ESP I'm currently using for testing, and it 'works on my machine' 😃

Sminter4, 2022-02-21:
Thanks for your guidance!
Now `system#boot` works again.

tonhuisman, 2022-02-21:
But what have you done to _get_ it working?

Sminter4, 2022-02-23:
Unfortunately I changed too many things to be able to say clearly.
I implemented both of your posts.
a) delete and reactivate rules.
b) Vatiable instead of Relay to visualize the result from `system#boot`
c) Query of the button changed. The button has "`1`" by default. The rule for switching the relay now queries the status "`0`" and no longer the "`1`". However, this shouldn't have any effect since the `bootstate` was disabled...

tonhuisman, 2022-02-23:
Well, the most important is that it is working as intended 👍  😃

tonhuisman, 2023-09-09:
This seems to be solved, so can be closed.

closed 2023-09-09 by TD-er