letscontrolit/ESPEasy #3942: PCF8574 can't be used for output anymore?

ghtester, 2022-02-02:
After some (long) time I wanted to play with the PCF8574 module but it looks in the meantime the appropriate plugin (P019) was changed to Switch Input only and despite there are still PCFGPIO etc. commands mentioned in documentation ( https://espeasy.readthedocs.io/en/latest/Plugin/P019.html ), it seems they don't work anymore (Failed).

Is there currently a way to use PCF8574 as output device except wdconfig command please?

TD-er, 2022-02-02:
Have you seen the [PCF GPIO commands](https://espeasy.readthedocs.io/en/latest/Reference/Command.html#external-pcfgpio) ?
The PCF and MCP GPIO expanders have been moved to be internal core components. (over 2 years ago)

ghtester, 2022-02-02:
Yeah that's what I mentioned above - the PCFGPIO commands returns Failed status, tested most of them but only wdconfig worked... Tested on Custom Vagrant build from 20220129.

TD-er, 2022-02-02:
Do you have P019 and P009 included in your build?

ghtester, 2022-02-02:
Only P019.  But maybe I am stupid and forget to use a proper device address... To be checked...

tonhuisman, 2022-02-02:
The pin number used determines the I2C address, afaik.

ghtester, 2022-02-02:
Yeah, the bad address... :-( I forgot to use a proper offset. I am sorry for the hoax, PCFGPIO works fine.

closed 2022-02-02 by ghtester

TD-er, 2022-02-02:
Glad it got "fixed" in 30 minutes :)

ghtester, 2022-02-02:
Shame on me for writing an issue too early - next time I'll test more precisely before writing hoaxes... ;-( Sorry again for taking your valuable time due to my stupidity...

TD-er, 2022-02-02:
Don't worry about it.