letscontrolit/ESPEasy #3943: [P036] request to left align the text

net1957, 2022-02-04:
### Checklist
- [x ] The title of this issue is "to the point" and descriptive.
- [x ] This issue describes what is happening.
- [x ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I use the configuration page to add text to be displayed. The text is centered

### Expected behavior

We should have an option to left align the text on a line

### Actual behavior

The lines are centered, so it's difficult to align values on different lines

### Steps to reproduce

1. type lines with different lengths on the configuration page 

version:
[ESP_Easy_mega_20211224_energy_ESP32_4M316k Dec 24 2021](https://github.com/letscontrolit/ESPEasy/releases)
[image]

The configuration page: (3 lines/page)
[image]

The result for page 2 ( lines 4-6)
[image]
same formatting for page 1

tonhuisman, 2022-02-06:
@net1957 
I've now added that as a feature, while working on optimizing the code for this plugin. (see linked PR above this message).
Please test, you can [download it from here](https://github.com/letscontrolit/ESPEasy/actions/runs/1803425872) (valid until the next commit in that PR 😉)

Be aware that all fonts used use variable width characters (proportional), so aligning values may be a bit of trial and error. Spaces on the left side of the content are left intact, when Left-align is active (for Centered content leading and trailing spaces are removed).

net1957, 2022-02-07:
Nice! I checked it and it's working as expected.
Regards

tonhuisman, 2022-02-11:
Updated the download link above, for if you want to test with the latest (speed) improvements in ESPEasy, as merged recently.

net1957, 2022-02-12:
I checked this version and didn't find any problem.

How can I verify the speed improvement?

tonhuisman, 2022-02-12:
> How can I verify the speed improvement?

Serving of pages from the ESP to the (webbrowser)client has much improved, mostly by all the work the TD-er put into that, but also the adoption of SDK 3.0.
The display will still work at the same speed, I'd expect 😄, even though I did some small optimizations, you probably won't be able to notice the difference.

TD-er, 2022-02-12:
I did tweak lots of string handling in the code.
Mainly when serving web pages, but also in other parts of the code.
So a lot of temporary memory allocations have been removed and we also gained some more free memory.

tonhuisman, 2022-02-12:
@net1957 I've changed the font used for the 4-lines display (again), this time to a non-proportional font but with better readability, can you re-test, please, to see if it still works as expected and intended?

net1957, 2022-02-12:
I checked it with (start  with line 1) :
````
Circuit 1: GENERAL
[pzem_1#P_1#d4]W     C:  [pzem_1#E_1#d4.2]KW/H
I: [pzem_1#A_1#d2.2]A   U:   [pzem_1#V_1#d3.2]V 
c: [pzem_1B#C_1#D.2]      F:     [pzem_1B#F_1#d2.1]HZ
Circuit 2: vers ONDULEUR
[pzem_2#P_2#d4]W     C:  [pzem_2#E_2#d4.2]KW/H
I: [pzem_2#A_2#d2.2]A   U:   [pzem_2#V_2#d3.2]V 
c: [pzem_2B#C_2#D.2]      F:     [pzem_2B#F_2#d2.1]HZ
````

The results are below.
the lines 2 and 6 (xxxW) are not correctly aligned (the first value is 3 or 4  char. length)

[image]
[image]
[image]

Regards

net1957, 2022-02-13:
Don't known if this is related, but the notification plugin seems missing. (build: ESP_Easy_mega_20220212_energy_ESP32_4M316k)

[image]

tonhuisman, 2022-02-13:
When hiding the Indicator, the next size font is used, and that's not mono-spaced, so vertical alignment isn't very good/reliable.

> the notification plugin seems missing

I don't think that the Notification plugins where ever in the Energy builds, to save some space in the .bin file. I think there currently is enough space available to include them.

I've created a PR to add them to the Energy builds.

ghtester, 2022-02-13:
> the notification plugin seems missing

I can confirm that Notifications are missing as well in my custom build made yesterday (despite I have E-mail and Buzzer enabled in the Custom.h). In the custom build from 20220129 it was OK.

net1957, 2022-02-13:
Thanks for the tips. The result:

[image]

It's aligned but a little to small... The previous test without the indicator is more readable. Could this case be mono-spaced?

I can confirm that in previous version the notification was enabled in the energy build (it was OK in 20211224 or later)

PR #3938 (2022-04-04): [P036] Bugfix: Re-enable 4-line frames, Feature: Left-align content
Resolves #3936 
Resolves #3943 

- By tweaking the height of the footer and the font metrics, the 4-line frames can now be displayed again.
- Bonus 1: Add sub-command `linecount` for changing the number of lines per frame. Activates on next Interval. (Disabled on TEST and MINIMAL builds)
- Bonus 2: Add setting to optionally hide the Indicator (dots showing page/active page in footer), so this space can be used for content, often using a larger font
- Bonus 3: Add Left-align option for content, including `leftalign` sub-command [... cut]
closed 2022-04-04 by TD-er