letscontrolit/ESPEasy #3948: ESP32 Password problem

fly74, 2022-02-12:
Hi Guys,

I face the same in `https://www.letscontrolit.com/forum/viewtopic.php?t=8935`. Is it still not possible to update to the actual release from 202110xxx and login after that?

TD-er, 2022-02-12:
It is fixed in the IDF4.4 PR: https://github.com/letscontrolit/ESPEasy/pull/3845
I am close to merging it, but I didn't want it to be merged very quickly after the SDK3.0.0 PR I merged this week.
Both PRs consist of over 100 commits, so if I messed up in one of them it is already hard to find where I messed up, let alone if multiple of these are merged.
Still it will be merged soon. Perhaps 1-2 weeks, with a build in between.

fly74, 2022-03-04:
@TD-er Any news?

closed 2022-04-27 by TD-er

fly74, 2022-04-27:
Confirmed fixed in ESP_Easy_mega_20220427_normal_ESP32_4M316k.bin

TD-er, 2022-04-27:
Just to be complete, if someone will find this issue.
It is fixed in the code merged today, after I made a new build.
The new build of course also has this date in the build name, but this fix is only present in official builds made after 20220427 or when someone makes his/her own builds.

fly74, 2022-04-27:
@TD-er is not fixed in the today release ?

TD-er, 2022-04-27:
> @TD-er is not fixed in the today release ?

If it is, then I would be a bit surprised as I was convinced it was related to something fixed upstream in the later SDK.
Did you test it with the latest build?

fly74, 2022-04-27:
> If it is, then I would be a bit surprised as I was convinced it was related to something fixed upstream in the later SDK.
> Did you test it with the latest build?

yes, with the release I wrote above.

PR #3845 (2022-04-27): [ESP32] Update to IDF 4.4 Arduino-ESP32 2.0.1
Including lots of WiFi fixes to be able to get a stable WiFi connection on the new ESP32 SDK.

This makes the bin file size quite a bit smaller (300k less on "Custom" build)
See this comment about the effects of the new bootloader in IDF 4.4: https://github.com/esphome/esp-web-tools/issues/151#issuecomment-1004296924

Fixes: #3860
Fixes: #3948

ToDo:

- [ ] Test I2C plugins  (see: https://github.com/espressif/arduino-esp32/pull/5683 & https://github.com/espressif/arduino-esp32/issues/6373#issuecomment-1059196867  & https://github.com/espressif/ardu [... cut]