letscontrolit/ESPEasy #3951: 64 programable WS2812-LEDs

fly74, 2022-02-14:
Is there a plugin for this kind of displays?

[WS2812-LEDs](https://www.az-delivery.de/products/u-64-led-panel?variant=40362432466&utm_source=google&utm_medium=cpc&utm_campaign=azd_de_google_shopping_standard_labelled-products&utm_content=132255586266&utm_term=&gclid=EAIaIQobChMInerTyc_-9QIV9IODBx3ueQLSEAQYASABEgKH5_D_BwE)

TD-er, 2022-02-14:
Yep, in [our plugins](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins), we call them "Neopixel"

fly74, 2022-02-14:
Oh, I give it a try.

closed 2022-02-14 by fly74

tonhuisman, 2022-02-15:
I wrote some documentation on the NeoPixel Basic plugin P038 for PR 3909, but that hasn't been merged yet, so I generated a pdf of that page:
[Output - NeoPixel (Basic) — ESP Easy 2.1-beta1 documentation.pdf](https://github.com/letscontrolit/ESPEasy/files/8074110/Output.-.NeoPixel.Basic.ESP.Easy.2.1-beta1.documentation.pdf)

For generic help on rules, you can go to the RTD documentation page on [rules](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html)

fly74, 2022-02-18:
@TD-er Nice as cascade [image]

TD-er, 2022-02-18:
Cool :)
I should also start playing with those.
I do have a few of them in my drawer-of-things-to-add-to-ESPEasy, but have not yet played with them.

fly74, 2022-02-18:
Is it only possible make some simple signs with NeoPixelLine,... or can we put an bitmap to variable and show it?

TD-er, 2022-02-18:
Not as far as I know, but like I said, I really need to start playing with it, so I find those shortcomings myself ;)

tonhuisman, 2022-02-18:
You can create a new issue requesting that, so it won't go out of sight.

fly74, 2022-02-18:
I make a new isssue.