letscontrolit/ESPEasy #3956: Strange Behaviour since commit 779ae24a79407fd33242dc80daf0a4105de75d7d

clumsy-stefan, 2022-02-16:
I see a strange behaviour after commit 779ae24a79407fd33242dc80daf0a4105de75d7d (PR #3946). One a node (with weak WiFi) up to mentioned commit everything seems fine but using the last commit 8c75f0959ecc35c819ed1413937e88d4b55e3049 Wifi, Load and Memory shows significant differences (see graphs below).

Updated from 779ae24a79407fd33242dc80daf0a4105de75d7d to 8c75f0959ecc35c819ed1413937e88d4b55e3049 around 17:00 yesterday and back to previous again today around 08:00.

It seems that function is not affected, but I wanted to report anyways. As I'm using ArduinoIDE the changes in the config files from PR #3953 most likely don't affect my build, only the .ino/.cpp/.h ones.

[image]
[image]
[image]

TD-er, 2022-02-16:
See also the discussion from this morning: https://github.com/letscontrolit/ESPEasy/pull/3953#issuecomment-1041581802
@ghtester also found something strange

clumsy-stefan, 2022-02-16:
Hmm... in my builds I neither included IR nor the serialserver plugin.... which seem to be the issue in #3953. I am currently a bit limited in time to nail down which commit generate this, but I can dig further over/after the weekend...

TD-er, 2022-02-16:
> Updated from [779ae24](https://github.com/letscontrolit/ESPEasy/commit/779ae24a79407fd33242dc80daf0a4105de75d7d) to [8c75f09](https://github.com/letscontrolit/ESPEasy/commit/8c75f0959ecc35c819ed1413937e88d4b55e3049) around 17:00 yesterday and back to previous again today around 08:00.

Looks like every commit made on Feb 14th...

clumsy-stefan, 2022-02-16:
yep, so in fact PR #3953, but which commit within that PR causes my issue I can't yet say... interestingly I only see it on this very node... (which in fact is one with the worst signal reception).

TD-er, 2022-02-16:
Do you have Arduino OTA included in your build?

clumsy-stefan, 2022-02-16:
> Do you have Arduino OTA included in your build?

Yes

>And if so, can you test on 1 node (easy to reach) if it makes a difference with ARDUINO_OTA undefined?

Now that's a problem, as I have updated all my ~30 nodes with this build and I see this behaviour only on exactly one node (though this is the only sonoff-basic unit), which is remote and difficult to get to, it's not that easy... I can't see the issue on any other node until now.

what I do see is that the latest commit (8c75f0959ecc35c819ed1413937e88d4b55e3049 PR #3953) seems to improve the situation (see graphs below) but not completely solve it... 

updated to 8c75f0959ecc35c819ed1413937e88d4b55e3049 around 12:00 today...

[image]
[image]
[image]

TD-er, 2022-02-16:
I have looked at the code changed on the 14th.
Not many things have changed. Most are just extra includes (some user keeps nagging me for those, so I wanted to beat him to it ;) )
There is one "significant" change, which is on the Arduino OTA.
This is a bit strange as how it was done.
Problem is that including the ArduinoOTA.h does define a global object.
Now I moved it to a single file where it is included, as you may see funky things when such global objects are defined more than once.
For example, it is theoretically possible you get multiple different objects with the same name.

The other "significant" change is the `UTF8_charLength` function.
But that should have been  a bug before.

clumsy-stefan, 2022-02-16:
what about 981af66379cbb2fa457a12e7da88d32c6ffaa993 and 7bc5e21d62477c1bf65da1d43d91aa9489773d1c? can they make a change?

My problem is I don't see any significant difference between this one node and all my others... so I don't understand why it happens only on this one (except, that it probably has one of the worse WiFi signals)... same tasks, same rules, same config... and when I browse the web-IF it's quick, responsive, no problems whatsoever... just the strange memory behaviour...

clumsy-stefan, 2022-02-16:
ok, thanks..

I did now find one difference between this and (all) other nodes, I had Eco mode disabled on this one and enabled on (all) others... Even though it seems strange to me that this makes a difference, I'll do some tests on that.

Still strange, that without changing ECO mode, you can see a different behaviour...

I'll report if I find something... but to be honest, I don't think that's a serious issue to use too much time for, just an observation to keep in mind if we run into a more serious that could relate or give hints from this..

TD-er, 2022-02-16:
The Eco mode does do some fishy things.
What I called "Eco mode" is nothing more than calling `delay()` for as long as there is nothing scheduled to do in the scheduler.
The result is that the WiFi radio in the ESP does enter some low-power mode after a while.
Sending a few pings from just any host in the network to that node will put the WiFi into max. power mode again for a while.
What it actually does, is it increases the DTIM interval.
The most responsive (and power intensive) DTIM is 1.
So it does listen for every beacon interval from the access point.
With a DTIM of 2, it will skip every other beacon interval (typically 102.4 msec) and thus allow for a longer sleep period.

clumsy-stefan, 2022-02-16:
but why should it have such an effect, without changing it... in this graph ECO mode was off just changed the build's and the memory got stable...
[image]

TD-er, 2022-02-16:
And there is no change between 1 heap or 2?

clumsy-stefan, 2022-02-16:
no, really just the last PR's included...

the unit is responsive, accessible OTA works fine... it's just the numbers... I'm doing new versions now... and some tests on ECO mode, but still, looks strange but ist' not really a problem...

clumsy-stefan, 2022-02-17:
Some feedback from tonight with the various versions. I can't see a real dependency though... except that turning ECO mode on seems to improve...

[image]

[image]

clumsy-stefan, 2022-02-23:
to be honest, I think we shoould close this issue, it's not really reproducable and for me not a serious issue...

closed 2022-02-23 by TD-er