letscontrolit/ESPEasy #3959: Rule problem

fly74, 2022-02-18:
I have a Rule like this
`On System#Boot do
     Event,poff
endon

On pon do
   NeoPixelAll,5,0,0
endon

On poff do
   NeoPixelAll,0,0,0
endon

On MS#State do
if [MS#State] = 1 then
    Event,pon
else
    Event,poff
endif
enddo`

but the log says:

`107974 : Info   : EVENT: MS#State=0
108008 : Error  : Calculate: Unknown token input: 1 then = 0
108018 : Info   : ACT  : Event,poff
108020 : Info   : EVENT: poff
108041 : Info   : ACT  : NeoPixelAll,0,0,0
112166 : Info   : SW  : GPIO=14 State=1 Output value=1
112186 : Info   : EVENT: MS#State=1
112228 : Error  : Calculate: Unknown token input: 1 then = 0
112238 : Info   : ACT  : Event,poff
112240 : Info   : EVENT: poff
112261 : Info   : ACT  : NeoPixelAll,0,0,0
114867 : Info   : SW  : GPIO=14 State=0 Output value=0
114886 : Info   : EVENT: MS#State=0
114928 : Error  : Calculate: Unknown token input: 1 then = 0
`

What means "Unknown token input"? Event from HTTP don't work too.

tonhuisman, 2022-02-18:
You should probably leave out the spaces around the `=`, and this script doesn't use `then` 😉 
The `enddo` is to be replaced by `endon`
Also an improvement, use:
`if %eventvalue1%=1`, as theoretically, the state could have changed between the event being fired, and the rule processed.

fly74, 2022-02-18:
Shame on me, you're right, works with event, but from http it don't work with curl http://host/control?cmd=event,pon

closed 2022-02-18 by fly74