letscontrolit/ESPEasy #3963: A slow (re)connect to hidden AP

ghtester, 2022-02-23:
I know the connection to hidden AP (with SSID broadcast disabled) is more complicated due to SDK limitation (as far as I can remember, this was already discussed here some time ago) so it takes significantly longer but it needs to be accepted.
Unfortunately the reconnection to hidden AP is also quite long (usually several minutes) despite the **Use Last Connected AP from RTC:** option in Advanced Settings is enabled. 
I suppose the **Last Connected AP's SSID** is stored to RTC and used for reconnection.
The question is - could it be improved somehow, for instance with saving BSSID to RTC and use this concrete BSSID for reconnection instead of SSID? 
I don't know if it's even possible but sometimes it would be nice to have a global option to configure AP BSSIDs instead of SSIDs in WiFi settings.

TD-er, 2022-02-23:
BSSID and channel and the index of the active WiFi key setting are stored in RTC.
So the reconnect should take only a few 100 msec, as long as the BSSID is reachable and the channel remains the same.
If this does not apply, then it will take quite some time to reconnect as it will perform some scan and try to reconnect to hidden SSID networks.
But still this should take only a few seconds, not minutes.
If it takes minutes, then I wonder whether this hidden SSID access point may be at the far end of what is reachable, or maybe the access point may be slow in responding to a connect attempt.
I know ESPEasy may not always wait long enough for an access point to reply to a connect request.

ghtester, 2022-02-23:
OK, thanks for the details. The AP model is really old, D-Link DWL-2100AP. The reconnect takes several seconds when I enable the BSSID broadcast on it.  Currently the ESP node is located next to the AP so tthe signal is strong, the same behaviour on some other ESP nodes located several meters from AP.
The AP list after Scan show about 23 APs around and the last one in the lis is #Hidden#  (=mine).
I 'll try with another AP model.

ghtester, 2022-02-25:
Tested with another (also a very old model) AP and reconnect takes several second with it... So it's an issue related to AP model... Closing the issue.

closed 2022-02-25 by ghtester

tonhuisman, 2022-02-25:
Wouldn't it be interesting to see if the ESP does a faster connect with a newer AP?

TD-er, 2022-02-25:
The latest builds (not sure when I changed it) already have a 'tweak' to make a more quick connecting to a hidden AP.
Older builds did check all combinations of hidden APs and passwords, but now when I cannot connect to a non-hidden AP, I try to connect to credentials nr 1 and then to nr 2.
Still this should not take very long.
Maybe you can uncheck the reconnect from RTC on the tools->Advanced page, as it changes how it does scan on a reconnect.

ghtester, 2022-02-27:
> Wouldn't it be interesting to see if the ESP does a faster connect with a newer AP?

I don't think it depends on AP's age but more likely on AP's features / parameters configurable. And trying to discover the root of the issue is probably just wasting the time. The most important is it connects in the end. 

> Maybe you can uncheck the reconnect from RTC on the tools->Advanced page, as it changes how it does scan on a reconnect.

I tried (not just) that settings as well, without any positive effect. It must be an unlucky specific ESP / AP combination but it's not so important when it works fine with different AP(s).

ghtester, 2022-03-08:
BTW.  I have replaced the AP with TP-LINK Archer MR200  and the ESP nodes usually can't connect anymore when I set the SSID hidden.  Also Lenovo tablet can't connect to this hidden SSID. Some other devices like Xiaomi phone has no issue with connection to hidden SSID at all... All firmware at latest available versions... So it looks the compatibility across WiFi devices is worse than it used to be...

ghtester, 2022-03-10:
That would be great, thank you in advance. 
I also believe the MQTT connection status should be separated from WiFi connection status. Because when the MQTT controller is down, it does not mean the WiFi is also down and the current connectivity test does not reflect that and it makes serious issue with (unnecessary) WiFi reconnection instead of just MQTT reconnect attempts.
Last but not least, perhaps the WiFi (re)connect process could be improved / simplified and the BSSID enabled in WiFi configuration. Please note the ESP node with 4 years old firmware is still connecting more reliably to WiFi APs. From my point of view the configurable variable could be added to both Primary and secondary AP configuration which should specify the connection attempts made to the AP and switch to another if not successful. And the scan for APs should be perhaps done only and one time when there's an initial configuration.
Another most important thing (but it's an OffTopic here) is a more RAM demand in latest firmware builds (at least in my Custom) which leads to often crashes so I am able to achieve from several days to maximum 1 month uptime only in comparison with old ESP which has 254 days uptime now... Yeah the new features and improvements are great but...