letscontrolit/ESPEasy #3966: [ESP32]

clumsy-stefan, 2022-03-01:
On my ESP32 Builds, when defining a status led in ' hardware' I immediately get the following error nonstop:
```
E (22894) ledc: ledc_get_duty(501): LEDC is not initialized
E (22899) ledc: ledc_get_duty(501): LEDC is not initialized
E (22904) ledc: ledc_get_duty(501): LEDC is not initialized
E (22909) ledc: ledc_get_duty(501): LEDC is not initialized
```

and the node slows down. I couldn't find anywhere in the code where this error is produced. Any Idea?

tonhuisman, 2022-03-01:
Dropped that error in Google and found this: https://forum.arduino.cc/t/esp32c3-errors-ledc/960275

clumsy-stefan, 2022-03-01:
Hmm... in `src/Helpers/Hardware.cpp` the setup is done according to that post you reference... at least if `ESP32` is defined (which obviously should be)... I'll dig deeper... thanks for the pointer..

TD-er, 2022-03-01:
Please also test with this PR: https://github.com/letscontrolit/ESPEasy/pull/3845

clumsy-stefan, 2022-03-01:
Thanks, I didn't see that PR..

 First quick test shows that:
- the web interface is significantly faster
- the ledc error is gone...

I'll do some more tests tomorrow, how it runs, so we can solve/delete this issue...

TD-er, 2022-03-01:
> I'll do some more tests tomorrow, how it runs, so we can solve/delete this issue...

And allow to merge that PR too, if it is running fine (and no other strange issues are found)

clumsy-stefan, 2022-03-02:
did new versions for esp8266 as well as for esp32 with the mentioned PR currently running for ~1h

the good:
- it does not seem to have any (negative) impact for the 8266/8285 builds
- it does seem to improve responsiveness of the esp32 (web ui).

the bad:
- pwm does not seem to work on the esp32, at least not on the builtin led.

I'll keep them running for some time now, if it stays like this I guess you could merge it, just need to look into the pwm stuff for esp32...

TD-er, 2022-03-02:
Can you show from the sysinfo page what the used SDK is?
In the PlatformIO configuration, I have a special prepared SDK, which is stripped from a lot of (debug) stuff we don't need, so the builds are significantly smaller compared to older builds.

clumsy-stefan, 2022-03-02:
I'm using latest GIT
[image]

clumsy-stefan, 2022-03-03:
All my esp8266 units run now for over 24h now without any issues. Also a few esp32 units run without problems with this PR.

So from my point of view it could be merged and then in a separate issue/PR dig deeper into the esp32 PWM (and possibly other) issue. IMHO..

clumsy-stefan, 2022-03-05:
I'm a step further on the LED issue. It seems there are 2 different issues which both lead to non working PWM on esp32.

First one is while initialising the channels in `src/Helpers/Hardware.cpp` line 1610:
```
ledChannelFreq[x] = ledcSetup(x, 0, 10); // Clear the channel
```

`ledcSetup()` throws an error when using a frequency of `0`. Changing this to another value (eg. `1000`) the `LEDC is not initialised` at unit startup is gone.

As a follow-up error, the first time pwm is set it throws again an error on line 1640:
```
if (static_cast<uint32_t>(ledcReadFreq(x)) == ledChannelFreq[x]) {
```
as `ledcReadFreq()` always fails with `LEDC is not initialised` and therefore `mustSetup` will always be false and no channel is set up. Changing the frequency in line 1610 when initialising also fixes this error.

The second issue is that `pinMode(pin, OUTPUT);` on line 1668 seems to break PWM setup again, but I have no clue why... This line can be commented/removed.

As this is also called when using the `GPIO -> LED` setting on the hardware page it has to be excluded in `src/Helpers/Network.cpp` lines 27-29:
```
  if (gnStatusValueCurrent < 0) {
#ifndef ESP32
    pinMode(Settings.Pin_status_led, OUTPUT);
#endif // ESP32
  }
```

So changing these 3 things make PWM work as intended on ESP32 (at least on my nodes). ;)

clumsy-stefan, 2022-03-05:
I guess I can't make a PR to another PR? At least I don't know how..

tonhuisman, 2022-03-05:
> I guess I can't make a PR to another PR? At least I don't know how..

I think you can pull that PR local, and push commits to it?

clumsy-stefan, 2022-03-05:
I don't have the permission to push to another/foreign PR...

tonhuisman, 2022-03-05:
Then we'll leave it to TD-er to handle it, it's just a few small changes 😉

clumsy-stefan, 2022-03-05:
Now that's strange... I recompiled and reflashed my nodes, now it doesn't work anymore... no code changes... 😒

TD-er, 2022-05-02:
Closed by: https://github.com/letscontrolit/ESPEasy/pull/4030

closed 2022-05-02 by TD-er