letscontrolit/ESPEasy #3967: New device support  AHT20 + BMP280 (2 in 1)

almirus, 2022-03-03:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

not show any Environment information 

### Expected behavior

Temperature
Humidity
Pressure

### Actual behavior

Temperature:0
Humidity:0
Pressure:0

### Steps to reproduce

1. Connect new device via i2c (power + SDA: GPIO-4 (D2) SCL: GPIO-5 (D1))

### System configuration

Hardware: 
wemos d1 mini and [new device AHT20 + BMP280](https://aliexpress.ru/item/1005003123605458.html?sku_id=12000024222833412&spm=a2g39.orderlist.0.0.28464aa6nkAWE2)

ESP Easy version: 
[ESP_Easy_mega_20211224_normal_ESP8266_4M1M Dec 24 2021](https://github.com/letscontrolit/ESPEasy/releases)

ESP Easy settings/screenshots: 
Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20211224_f162ebf
Plugin Count:⋄
47 [Normal]
Build Time:⋄
Dec 24 2021 14:39:35
Binary Filename:⋄
ESP_Easy_mega_20211224_normal_ESP8266_4M1M
Build Platform:⋄
Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20211224_f162ebf

### Rules or log data
```
298461: EVENT: Clock#Time=Thu,21:24
298465: BMx280 : Unable to detect chip ID (255, failed)
298465: BMx280 : Unable to detect chip ID (255, failed)
299460: BMx280 : Unable to detect chip ID (255, failed)
299460: BMx280 : Unable to detect chip ID (255, failed)
300460: BMx280 : Unable to detect chip ID (255, failed)
300460: BMx280 : Detected BMP280
301460: BMx280 : Unable to detect chip ID (255, failed)
301460: BMx280 : Unable to detect chip ID (255, failed)
```
[image]

tonhuisman, 2022-03-03:
Have you tried to use the sensors separately? And does it then work as intended?
How are the sensors wired, as in: length of the wires?, placement/location of pull-up resistors on the I2C bus?

TD-er, 2022-03-03:
See the warning on the [documentation page of the ATH10/20/21](https://espeasy.readthedocs.io/en/latest/Plugin/P105.html#description)
The sensor does not comply with I2C standard and cannot be used with any other I2C device on the same I2C bus.

tonhuisman, 2022-03-03:
I think that warning only applies to AHT10/AHT15. The AHT20/AHT21 are of an updated design, and _should_ work fine, but you'll never know until actually tested...

almirus, 2022-03-03:
> Have you tried to use the sensors separately?
its not possible 
this device have 2 sensors ([AHT**20** + BMP280](https://aliexpress.ru/item/1005003123605458.html?sku_id=12000024222833412&spm=a2g39.orderlist.0.0.28464aa6nkAWE2)) in one
[image]

> See the warning on the [documentation page of the ATH10/20/21](https://espeasy.readthedocs.io/en/latest/Plugin/P105.html#description)

The problem is known only for this device AHT10. I have AHT20

tonhuisman, 2022-03-03:
And the wiring (length & quality of soldering)? Just pushing some wires into the holes of the board isn't going to win any prices.
You can also try the low speed I2C checkbox (100 kHz instead of 400 kHz, per default I2C configuration).

Pull-up resistors seem to be on-board, so that should be fine.

almirus, 2022-03-03:
> And the wiring (length & quality of soldering)?

[image]

tonhuisman, 2022-03-03:
> and Standard firmware doesn't support AHT*** sensors

You can install the TEST_A .bin file, that does include the AHTxx sensors, next to everything that is in the NORMAL .bin.

Wiring seems to be fine, you have powered it from 3V3, I assume?

almirus, 2022-03-03:
> Wiring seems to be fine, you have powered it from 3V3, I assume?
yep

> You can also try the low speed I2C checkbox
🍻 it's work!!!
[image]

almirus, 2022-03-03:
> Quite cold where you are now, 0.00? 😺

no, temperature fluctuates from 0 to 25 😌

`

2341468: BMx280 : Unable to detect chip ID (0)
2342023: WD   : Uptime 39 ConnectFailures 0 FreeMem 18488 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
2342468: BMx280 : Unable to detect chip ID (0)
2343468: BMx280 : Unable to detect chip ID (0)
2344468: BMx280 : Unable to detect chip ID (0)
2345468: BMx280 : Unable to detect chip ID (0)
2346468: BMx280 : Unable to detect chip ID (0)
2347468: BMx280 : Detected BMP280
2348468: BMx280 : Unable to detect chip ID (0)
2349468: BMx280 : Unable to detect chip ID (0)
2349483: Unknown : Address: 0x77
2349486: Unknown : Temperature: 25.48
2349488: Unknown : Barometric Pressure: 718.99
2349498: EVENT: temperature#Temperature=25.48
2349598: EVENT: temperature#Humidity=0.00
2349697: EVENT: temperature#Pressure=718.99
2350468: BMx280 : Unable to detect chip ID (0)
2351468: BMx280 : Unable to detect chip ID (0)
2352467: BMx280 : Unable to detect chip ID (255, failed)
2352468: BMx280 : Detected BMP280
2353468: BMx280 : Unable to detect chip ID (0)
2354467: BMx280 : Unable to detect chip ID (255, failed)
2354468: BMx280 : Detected BMP280
2355468: BMx280 : Unable to detect chip ID (0)

`

tonhuisman, 2022-03-03:
Strange that it detects a BMP280, but still shows a value for Pressure, as that isn't available in that sensor, according to the manufacturer specs 🤔 

I'd update to the TEST_A build, to see if the AHT20 sensor shows more reliable results.

almirus, 2022-03-03:
> I'd update to the TEST_A build

[image]

<pre>
152212: BMx280 : Unable to detect chip ID (0)
153212: BMx280 : Unable to detect chip ID (0)
154258: BMx280 : Unable to detect chip ID (0)
154261: AHT20 : reset
155212: BMx280 : Unable to detect chip ID (0)
155213: AHT20 : initialized
156211: BMx280 : Detected BMP280
157212: BMx280 : Unable to detect chip ID (0)
158212: BMx280 : Unable to detect chip ID (0)
158216: AHT20 : reset
158226: Unknown : Address: 0x77
158227: Unknown : Temperature: 25.48
158228: Unknown : Barometric Pressure: 718.99
158288: EVENT: temperature#Temperature=25.48
158388: EVENT: temperature#Humidity=0.00
158488: EVENT: temperature#Pressure=718.99
159211: EVENT: Clock#Time=Thu,23:06
159215: BMx280 : Unable to detect chip ID (0)
159218: AHT20 : unable to initialize
160212: BMx280 : Unable to detect chip ID (0)
160213: AHT20 : unable to initialize
161211: BMx280 : Unable to detect chip ID (0)
161213: AHT20 : unable to initialize
162344: BMx280 : Unable to detect chip ID (0)
162345: AHT20 : unable to initialize
163212: BMx280 : Unable to detect chip ID (0)
163213: AHT20 : unable to initialize
164211: BMx280 : Unable to detect chip ID (0)
164213: AHT20 : unable to initialize
165212: BMx280 : Unable to detect chip ID (0)
165213: AHT20 : unable to initialize
166212: BMx280 : Unable to detect chip ID (0)
166213: AHT20 : unable to initialize
166975: WD   : Uptime 3 ConnectFailures 0 FreeMem 15648 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
167212: BMx280 : Unable to detect chip ID (0)
167213: AHT20 : unable to initialize
168211: BMx280 : Unable to detect chip ID (0)
168214: AHT20 : initialized
169212: BMx280 : Unable to detect chip ID (0)
170211: BMx280 : Detected BMP280
170239: AHT20 : Addr: 0x38
170240: AHT20 : Temperature: -50.00 : Humidity: 3.13
170290: EVENT: aht20#Temperature=-50.00
170390: EVENT: aht20#Humidity=3.13
171326: BMx280 : Unable to detect chip ID (0)
172211: BMx280 : Unable to detect chip ID (0)
172228: Unknown : Address: 0x77
172229: Unknown : Temperature: 25.48
172229: Unknown : Barometric Pressure: 718.99
172288: EVENT: temperature#Temperature=25.48
172388: EVENT: temperature#Humidity=0.00
172488: EVENT: temperature#Pressure=718.99
173220: BMx280 : Unable to detect chip ID (0)
174223: BMx280 : Unable to detect chip ID (0)
175212: BMx280 : Unable to detect chip ID (0)
176230: BMx280 : Unable to detect chip ID (0)
177212: BMx280 : Unable to detect chip ID (0)
178272: BMx280 : Unable to detect chip ID (0)
179211: BMx280 : Unable to detect chip ID (0)
180211: BMx280 : Unable to detect chip ID (0)
181211: BMx280 : Unable to detect chip ID (0)
182280: BMx280 : Unable to detect chip ID (0)
182282: AHT20 : reset
183212: BMx280 : Unable to detect chip ID (0)
183213: AHT20 : initialized
184343: BMx280 : Detected BMP280
185212: BMx280 : Unable to detect chip ID (0)
186219: AHT20 : reset
186229: Unknown : Address: 0x77
186230: Unknown : Temperature: 25.48
186231: Unknown : Barometric Pressure: 718.99
186288: EVENT: temperature#Temperature=25.48
186388: EVENT: temperature#Humidity=0.00
186488: EVENT: temperature#Pressure=718.99
187211: BMx280 : Unable to detect chip ID (0)
187212: AHT20 : unable to initialize
188317: BMx280 : Unable to detect chip ID (0)
188318: AHT20 : unable to initialize
189212: BMx280 : Unable to detect chip ID (0)
189213: AHT20 : unable to initialize
190212: BMx280 : Unable to detect chip ID (0)
190213: AHT20 : unable to initialize

</pre>

tonhuisman, 2022-03-03:
Hmm, sounds like a really crappy combination, or very poor quality of chips, on that board 😞 

If you disable the task for BMP280, does that help in getting results from AHT20?

tonhuisman, 2022-03-03:
Hm, you installed TEST_A_beta, that has a later Arduino library version included. Next time, better pick the regular TEST_A build. But I don't expect that to make a difference in this release of ESPEasy though (newer releases _do_ have some differences in the beta build)

almirus, 2022-03-03:
after disable bmp280 🙈

[image]

<pre>
57700: AHT20 : initialized
59814: AHT20 : reset
60701: EVENT: Clock#Time=Thu,00:01
60705: AHT20 : unable to initialize
61700: AHT20 : unable to initialize
62256: WD   : Uptime 1 ConnectFailures 0 FreeMem 16248 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
62700: AHT20 : unable to initialize
63700: AHT20 : unable to initialize
64700: AHT20 : unable to initialize
65700: AHT20 : unable to initialize
66754: NTP  : NTP replied: delay 50 mSec Accuracy increased by 0.369 seconds
66755: Time set to 1646338330.369
66757: Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 180 min
66760: Local time: 2022-03-03 23:12:10
66761: EVENT: Clock#Time=Thu,23:12
66765: AHT20 : unable to initialize
66768: EVENT: Time#Initialized
67701: AHT20 : unable to initialize
68731: AHT20 : unable to initialize
69701: AHT20 : initialized
72701: AHT20 : reset
73702: AHT20 : unable to initialize
74701: AHT20 : unable to initialize
75700: AHT20 : unable to initialize
</pre>

TD-er, 2022-03-03:
The BMP280 you're using is not using the default I2C address.
Can you try to run it on the other address too?
If that does also somewhat work, like you're seeing now, then the sensor is not really set to a proper I2C address and hopping between addresses.

almirus, 2022-03-03:
> you installed TEST_A_beta, that has a later Arduino library version included. Next time, better pick the regular TEST_A build

same, with disabled bmp280
<pre>
58245: AHT20 : unable to initialize
59245: AHT20 : unable to initialize
60245: AHT20 : unable to initialize
61245: AHT20 : unable to initialize
62245: AHT20 : unable to initialize
63266: AHT20 : unable to initialize
64245: EVENT: Clock#Time=Thu,23:21
64249: AHT20 : initialized
66246: AHT20 : reset
67245: AHT20 : unable to initialize
68245: AHT20 : unable to initialize
69245: AHT20 : unable to initialize
70245: AHT20 : unable to initialize
71245: AHT20 : unable to initialize
72245: AHT20 : unable to initialize
73245: AHT20 : unable to initialize
74244: AHT20 : unable to initialize
75245: AHT20 : unable to initialize
76245: AHT20 : unable to initialize
76800: WD   : Uptime 1 ConnectFailures 0 FreeMem 19288 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
77245: AHT20 : unable to initialize
78246: AHT20 : initialized
81245: AHT20 : reset
82245: AHT20 : unable to initialize
83245: AHT20 : unable to initialize
84245: AHT20 : unable to initialize
85245: AHT20 : unable to initialize
86245: AHT20 : unable to initialize
87245: AHT20 : unable to initialize
88244: AHT20 : unable to initialize
89245: AHT20 : unable to initialize
90245: AHT20 : unable to initialize
91245: AHT20 : unable to initialize
92245: AHT20 : initialized
95282: AHT20 : reset
96245: AHT20 : unable to initialize
97245: AHT20 : unable to initialize

</pre>

but Humidity is 0

with enabled mpp280
<pre>
153245: BMx280 : Unable to detect chip ID (255, failed)
153246: BMx280 : Unable to detect chip ID (255, failed)
153247: AHT20 : unable to initialize
154245: BMx280 : Unable to detect chip ID (255, failed)
154246: BMx280 : Unable to detect chip ID (255, failed)
154247: AHT20 : unable to initialize
155351: BMx280 : Unable to detect chip ID (255, failed)
155352: BMx280 : Unable to detect chip ID (255, failed)
155355: AHT20 : unable to initialize
156245: BMx280 : Unable to detect chip ID (255, failed)
156246: BMx280 : Unable to detect chip ID (255, failed)
156247: AHT20 : unable to initialize
157247: BMx280 : Unable to detect chip ID (255, failed)
157248: BMx280 : Unable to detect chip ID (255, failed)
157249: AHT20 : unable to initialize
158245: BMx280 : Unable to detect chip ID (255)
158246: BMx280 : Unable to detect chip ID (255, failed)
158247: AHT20 : unable to initialize
159245: BMx280 : Unable to detect chip ID (255, failed)
159246: BMx280 : Unable to detect chip ID (255, failed)
159247: AHT20 : initialized
160245: BMx280 : Detected BMP280
[... 101 lines cut ...]
196800: WD   : Uptime 3 ConnectFailures 0 FreeMem 19056 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
197246: BMx280 : Unable to detect chip ID (255, failed)
197247: BMx280 : Unable to detect chip ID (255, failed)
198245: BMx280 : Unable to detect chip ID (255, failed)
198246: BMx280 : Unable to detect chip ID (255, failed)
198247: AHT20 : reset
199245: BMx280 : Unable to detect chip ID (255, failed)
199246: BMx280 : Unable to detect chip ID (255, failed)
199247: AHT20 : unable to initialize
200245: BMx280 : Unable to detect chip ID (255, failed)
200246: BMx280 : Unable to detect chip ID (255, failed)
200247: AHT20 : unable to initialize
201245: BMx280 : Unable to detect chip ID (255, failed)
201246: BMx280 : Unable to detect chip ID (255, failed)
201247: AHT20 : unable to initialize
202276: BMx280 : Unable to detect chip ID (255, failed)
202277: BMx280 : Unable to detect chip ID (255, failed)
202278: AHT20 : unable to initialize
203245: BMx280 : Unable to detect chip ID (255)
203246: BMx280 : Unable to detect chip ID (255, failed)
203247: AHT20 : unable to initialize

</pre>

 Humidity is 3.13

almirus, 2022-03-03:
> Can you try to run it on the other address too?

[image]
[image]
[image]

almirus, 2022-03-03:
> Can you try to run it on the other address too?

if i select an address 0x76(118) i get zero 
Temperature:0.00
Humidity:0.00
Pressure:0.00

tonhuisman, 2022-03-03:
The BMP280 might also need the Low I2C speed setting.

Are the wemos and sensor located close to the base of the lamp you have in the photo, above? The neon light might have a transformer in there that is causing interference on the ESP, sensor or wiring, so I'd place it at least half a meter away, and preferably more, just to avoid anything.

almirus, 2022-03-03:
> Are the wemos and sensor located close to the base

No, it was made for beauty 😇

almirus, 2022-03-03:
I noticed that the values 
[image]

 don't change 😔

tonhuisman, 2022-03-03:
In your screenshot the dust sensor isn't enabled, and both other sensors don't report normal readings, so probably nothing to update.
That Devices page requires to allow javascript to run on the page, I assume that's enabled/allowed?

almirus, 2022-03-03:
> That Devices page requires to allow javascript to run on the page, I assume that's enabled/allowed?

it's ok
[image]

tonhuisman, 2022-03-03:
If available, you could try another Wemos and set of wires, just to try something different. It looks like there is some interfering factor, not determined yet.

TD-er, 2022-03-03:
If the BMP does report in the logs that it cannot read the sensor ID, then that's where we should focus.

Most of the suggestions are already mentioned, but just to be complete:
- Cable length: max a few decimeters
- pull-up resistors of roughly 4k7 ... 10k must be present on the SDA and SCL pins, but sometimes those sensor boards have these too and when connecting multiple boards the effective pull-up resistance can become too low. -> Test with 1 board to see if that's working.
- Lower I2C clock frequency: Low speed checked on only 1 task may still cause issues on another I2C sensor which really doesn't like higher clock speeds -> Try setting both high and low speed on the hardware tab to 100 kHz
- The BMP280 and BME280 plugin (BMx280) is very stable, and tested extensively as those sensors are very popular. So if running with only that sensor connected is causing these errors too, then I would suggest to look at the board itself. e.g. check the pads for setting an address to see if any is populated or maybe a loose solder pad on that sensor board. Dirt may also cause issues.
- Check the voltage on the sensor(s). If this is not really a stable voltage, then the sensor may also act unpredictable.

TD-er, 2022-03-03:
One more thing you can try....
Maybe the ATH20 does need some clock stretching.
You can set it to something like "2000" on the tools->Advanced page.

almirus, 2022-03-07:
Hi guy's @tonhuisman & @TD-er thank you so much

today I cut the wire, shortened it (about 3cm) 

everything worked!

[image]

my log now
<pre>
300590: BMP280 : Address: 0x77
300591: BMP280 : Temperature: 25.06
300592: BMP280 : Barometric Pressure: 982.80
300646: EVENT: temperature#Temperature=25.06
300746: EVENT: temperature#Humidity=0.00
300882: EVENT: temperature#Pressure=982.80
302134: WD : Uptime 5 ConnectFailures 0 FreeMem 18808 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
308589: AHT20 : Addr: 0x38
308591: AHT20 : Temperature: 24.98 : Humidity: 31.82
308646: EVENT: aht20#Temperature=24.98
308746: EVENT: aht20#Humidity=31.82
311579: EVENT: Clock#Time=Mon,17:12
312590: BMP280 : Address: 0x77
312593: BMP280 : Temperature: 25.01
312594: BMP280 : Barometric Pressure: 982.82
312646: EVENT: temperature#Temperature=25.01
312746: EVENT: temperature#Humidity=0.00
312846: EVENT: temperature#Pressure=982.82
319590: AHT20 : Addr: 0x38
319592: AHT20 : Temperature: 24.98 : Humidity: 32.04
319646: EVENT: aht20#Temperature=24.98
319746: EVENT: aht20#Humidity=32.04
324590: BMP280 : Address: 0x77
324591: BMP280 : Temperature: 24.99
324592: BMP280 : Barometric Pressure: 982.79
324646: EVENT: temperature#Temperature=24.99
324746: EVENT: temperature#Humidity=0.00
324846: EVENT: temperature#Pressure=982.79
</pre>

tonhuisman, 2022-03-07:
Ah, great, then I suspect that the wires where acting like antennas picking up too much interference for I2C communication to work properly.

TD-er, 2022-03-08:
Great to hear it is working now.

closed 2022-03-08 by TD-er