letscontrolit/ESPEasy #3969: [ESP32] Bootloop/Crash when using DeepSleep

clumsy-stefan, 2022-03-05:
Just found an issue when using DeepSleep on an ESP 32:

when enabling sleep mode it goes to an infinite boot loop:
```
INIT : Booting version: SMY_13.5T4 Mar  2 2022 11:40:42 (ESP32 SDK v4.4-263-g000d3823bb)
1016 : Info   : INIT : Free RAM:258384
1017 : Info   : INIT : Soft Reboot #4 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
1018 : Info   : FS   : Mounting...
1885 : Info   : FS   : Mount successful, used 137297 bytes of 9209441
1922 : Info   : CRC  : SecuritySettings CRC   ...OK 
1926 : Info   : INIT : I2C
1928 : Info   : INIT : SPI not enabled
1929 : Info   : WiFi : Start network scan all channels
2038 : Info   : WiFi : STA Started
6745 : Info   : WiFi : Scan finished, found: 8
6748 : Info   : WiFi : Best AP candidate: clumsy_ap2 E4:8D:8C:49:4F:BA Ch:9 (-55dBm) WPA/WPA2/PSK
6751 : Info   : WIFI : Connecting clumsy_ap2 E4:8D:8C:49:4F:BA Ch:9 (-55dBm) WPA/WPA2/PSK attempt #0
6755 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
9313 : Info   : INIT : Free RAM:222048
13236 : Info   : INFO : Plugins: 25 [16M OTA SMY] 24tasks (ESP32 SDK v4.4-263-g000d3823bb)
13238 : Info   : EVENT: System#NoSleep=60
13604 : Info   : EVENT: System#Wake
13607 : Info   : EVENT: System#BootMode=1,1,0,0

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
```
Anyone used DeepSleep successfully?

tonhuisman, 2022-03-05:
This is when building for ESP32 with IDF 4.4, I assume?

clumsy-stefan, 2022-03-05:
yes, sorry, forgot to mention, also with PR #3845 included.

I guess PSRAM is not too well supportet yet, as I read in various places...

TD-er, 2022-03-05:
Hmm this is something that's fixed when using the exact SDK files linked in PlatformIO as I do use the SDK zipped by @jason8266
No idea how you can load those in Arduino IDE.

Jason2866, 2022-03-05:
Did i understand it correct there is no issue using my provided build with Platformio?
Which version do you use with Arduino? Core 2.0.2. This build is too old. Still have a lot of issues. There is no easy fast way to use my Platformio build with Arduino IDE.
A good time to switch to Platformio :-)

clumsy-stefan, 2022-03-05:
See log above, I use latest GIT build of IDF 4.4 for ESP32 (SDK v4.4-263-g000d3823bb) and latest GIT of ESP8266, so I guess it can't be too old...

TD-er, 2022-03-05:
> Did i understand it correct there is no issue using my provided build with Platformio?

I have to double-check it, but as far as I know, I have a test unit running the IDF4.4 code using PSRAM support and deepsleep (no PSRAM present on that board, but support is enabled on that build)

Jason2866, 2022-03-05:
Please try https://github.com/tasmota/esp-idf/tree/v4.4_dev
and 
https://github.com/tasmota/arduino-esp32/releases/download/2.0.2.3/framework-arduinoespressif32-release_IDF4.4.tar.gz
which you have to unpack and overwrite your existing Arduino version

Jason2866, 2022-03-05:
@TD-er btw Onewire needs changes since GPIO handling is massive refactored.
Using now IDF functions / calls. See https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreaker

clumsy-stefan, 2022-03-05:
> > Did i understand it correct there is no issue using my provided build with Platformio?
> 
> I have to double-check it, but as far as I know, I have a test unit running the IDF4.4 code using PSRAM support and deepsleep (no PSRAM present on that board, but support is enabled on that build)

Yes, if it's built but not available it's also no problem, only when built and actually also available it crashes!

clumsy-stefan, 2022-03-05:
> Please try https://github.com/tasmota/esp-idf/tree/v4.4_dev and https://github.com/tasmota/arduino-esp32/releases/download/2.0.2.3/framework-arduinoespressif32-release_IDF4.4.tar.gz which you have to unpack and overwrite your existing Arduino version @clumsy-stefan

Thanks, but what's the actual difference to the latest arduino-esp32 GIT?

As I'm going a few days abroad tomorrow, I can't really test anymore until next weekend, sorry...

clumsy-stefan, 2022-03-05:
> A good time to switch to Platformio :-)

PS: why does everyone want to get me off ArduinoIDE?? 😬 😀

TD-er, 2022-03-05:
> > A good time to switch to Platformio :-)
> 
> PS: why does everyone want to get me off ArduinoIDE?? 😬 😀

Since we've all been there and after getting to know platformio, still don't know why it took so long to start using it.

clumsy-stefan, 2022-03-05:
I think I'm just too lazy... as long as everything works, why should I take the pain in adapting all my predefined automations and merge them to PIO...

uzi18, 2022-03-05:
@TD-er In fact platformio is so elastic - it is hard do take step back :)

@Jason2866 there is compatible OneWireNg ...

@clumsy-stefan don't fear it, just create issue with your problems. In fact actually don't need to install anything on computer, can modify EspEasy and build firmware bin inside ... browser with gitpod.

TD-er, 2022-03-05:
> @TD-er btw Onewire needs changes since GPIO handling is massive refactored. Using now IDF functions / calls. See https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreaker

Not sure if ESPEasy needs to change as the 1wire plugins we use are bare metal bitbanging which I did tweak and tune using logic analyzers and protocol debuggers.
But good to know that one needs testing too....

clumsy-stefan, 2022-03-05:
> don't fear it

I don't fear it, I know PIO, as said I'm just too lazy... I'm using that many different workflows, frameworks, languages, protocols and systems (professionally and privately), I just don't change if I'm not forced to... keep it simple and never touch a running system  😆

Jason2866, 2022-03-05:
@uzi18 OneWireNG Not working with latest Arduino core and all ESP32 Mcu (S3)

uzi18, 2022-03-05:
@Jason2866 did you reported it?

Jason2866, 2022-03-05:
Have you tried all S3 Gpios? I had issues when i tried. Maybe it works now, since Arduino had changes too. Anyway the changed onewire driver works perfect and has a smaller footprint.

clumsy-stefan, 2022-03-19:
there seems to be a (possibly unrelated) other issue with esp32, I have units running without issues, but the log tells me:
```
WD : Uptime 84 ConnectFailures 0 FreeMem 199304 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: DISCONNECTED
```
even though the unit is perfectly reachable. Even the log is from the webpage.

Also the WebUI is really slow and laggy, probably becuase it thinks its disconnected? The UI on the ESP8266 is way faster...

clumsy-stefan, 2022-04-09:
I'm a bit further on this issue. It seems the ESP crashes in function `isDeepSleepEnabled()` when checking for PIN 16. On my module PIN 16 is not connected so I don't see why it crashes, but when excluding the `pinMode()` command:
```
#ifndef ESP32
  pinMode(16, INPUT_PULLUP);
#endif
```
for PIN 16 it won't crash anymore on the `nosleep` command....

Probably only a workaround, but at least it works.

Should I do a PR?

TD-er, 2022-04-09:
> Should I do a PR?

Yes please

TD-er, 2022-04-09:
Maybe we should look at other parts of the code too, where direct calls to pinMode etc. are done, as that pin is used on ESP32 by the PSRAM, or Ethernet in some setups.

clumsy-stefan, 2022-04-09:
yes that could obviously be... up to now I don't have any other issues on 4 different esp32 modules.

is PSRAM currently of any use in ESPEasy at all?

TD-er, 2022-04-09:
> yes that could obviously be... up to now I don't have any other issues on 4 different esp32 modules.
> 
> is PSRAM currently of any use in ESPEasy at all?

If present, it can be used... however I don't actually know if all memory allocations will be done on the PSRAM.
I know memory allocated in PSRAM cannot be used in all circomstances, like in interrupt handlers for example. (the same as when you need to mark stuff via IRAM attributes)

clumsy-stefan, 2022-04-09:
ok..

I guess this issue can be closed then (if you choose to include the PR/fix), as all other things discussed in here are not related to the original issue raised about the crash (especially as it has nothing to do with PIO or ArduinoIDE 😆).

PR #4002 (2022-04-09): [ESP32] pinMode() crashes ESP32 with PSRAM
pinMode() crashes the ESP33 when PSRAM is enabled and available. So don't check for ESP32.

Fixes: #3969
closed 2022-04-09 by TD-er