letscontrolit/ESPEasy #3973: activa/desactivar un relemmpm

adrianesp32, 2022-03-20:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

adrianesp32, 2022-03-20:
hola soy nuevo en esto de esp y programacion. actualmente tengo mi mcu8266 + modulo rele + firmware esp easy + app inventor, todo funciona de maravilla en mi red local desde el celular, el problema es que no se como conectarle un pulsador swicth a una entrada GPIO para mandarle el pulso y poder apagar o encender el rele, tengo conocimiento básico en programación, para el caso del swicth pulsador manual tengo que agregar como input ejemplo GPIO-14 (D5) y el rele lo tengo como salida en el GPIO-16(D0), pero al pulsar no hace ningun cambio el rele, estoy muy confundido en la logica de esp easy, porque yo le envio un pulso al GPIO-16 (D0) y no entiendo como se entera el GPIO-14(D5) que le estoy mandando un pulso????. ojala me puedan ayudar, porque todo es muy lindo con el celular, pero que pasa si me roban dicho celular, como enciendo la luz de mi casa???? se nesesita un pulsador manual para poder encender las luces del hogar.

adrianesp32, 2022-03-20:
[image]

tonhuisman, 2022-03-20:
(Translated using Google, common (spoken) language in this repository is English, so please try to use that in communication)

> Hi, I'm new to esp and programming. I currently have my mcu8266 + relay module + firmware esp easy + app inventor, everything works great on my local network from my cell phone, the problem is that I don't know how to connect a switch button to a GPIO input to send it the pulse and be able to turn off or turn on the relay, I have basic knowledge in programming, in the case of the manual pushbutton switch I have to add example GPIO-14 (D5) as input and I have the relay as output in GPIO-16(D0), but when pressed it does not no change in the relay, I am very confused in the logic of esp easy, because I sent a pulse to the GPIO-16 (D0) and I do not understand how the GPIO-14 (D5) finds out that I am sending it a pulse??? ?. I hope you can help me, because everything is very nice with the cell phone, but what happens if that cell phone is stolen, how do I turn on the light in my house???? a manual button is needed to turn on the lights in the home. 

> I leave an image for you to see and tell me what I'm doing wrong. 

Not much is wrong, except for maybe some more beginner oriented documentation.

Lets get this going, by following these steps:
- Open the Tools (wrench icon)/Advanced page
- Enable the option 'Rules'
- Enable the option 'Allow TaskValueSet on all plugins' (around 3/5 down the page)
- Submit the page (button near the end of the page)
- Open the Rules tab (appears between the plug/Devices and the Envelope/Notifications tabs)
- In the Rules 1 file add this code:
```
on switch#state=0 do // Event handler: when switch is closed (momentary button assumed, for on/off switch, remove the `=0` in this line)
  let,1,![door#state]  // Invert current relay state into variable 1
  TaskValueSet,door,State,[var#1] // Set the new state to the relay from variable 1
endon
```
- Save the rules
- Test 😃 

NB: As suggested in the issue template, this is more like a support-request, and should be handled in [the forum](https://www.letscontrolit.com/forum)

closed 2022-03-20 by adrianesp32

tonhuisman, 2022-03-21:
Received a personal response from the OP (translated using Google translate, removed some identifying stuff):
> I thank you for the help in controlling the 4-channel relay module with 4 switch buttons.
> 
> I did everything you told me and it worked perfectly, thank you much, I leave you in how the rule was.
> and then I'm going to bother in the forum so that they help me to be able to control 16 buttons (switch), to control 16 relays > (relay module 16) using the pwm controller pca9685, which only uses 2 GPIO SCL and SDA. Thank you very much.
>
> ```
> on switch2#state=0 do // Event handler: when switch is closed (momentary button assumed, for on/off switch, remove the  '=0' in this line)
>    let,1,![door2#state]  // Invert current relay state into variable 1
>    gpio,12,[var#1] // Set the new state to the relay GPIO pin from variable 1
> endon
> 
> 
> on switch#state=0 do // Event handler: when switch is closed (momentary button assumed, for on/off switch, remove the `=0` in this line)
>    let,1,![door#state]  // Invert current relay state into variable 1
>    gpio,16,[var#1] // Set the new state to the relay GPIO pin from variable 1
> endon
> 
> 
> on llave3#state=0 do // Event handler: when switch is closed (momentary button assumed, for on/off switch, remove the `=0` in this line)
>    let,1,![puerta3#state]  // Invert current relay state into variable 1
>    gpio,0,[var#1] // Set the new state to the relay GPIO pin from variable 1
> endon
> 
> on llave4#state=0 do //
>    let,1,![puerta4#state]  //
>    gpio,15,[var#1] //
> endon
> ```