letscontrolit/ESPEasy #3975: controlar 16 relay con pca9685

adrianesp32, 2022-03-23:
hola soy de argentina buenos aires, en mis pasa tiempos estoy modernizando mi casa inteligente, les comento que soy aprendis.
ya eh hecho funcionar exitosamente el modulo relay de 4ch + mcu8266 + firmware esp easy + app inventor y con la ayuda de ustedes le coloque 4 pulsadores  switch para apagarlos y encenderlo manaualmente, utilizando la regla. 
ahora lo que quiero es usar un modulo relay de 16 canales y utilice el modulo pca9685 para controlar los 16 relay, y mediante el navegador web funciona perfecto el control de los relay, la pregunta del millon es, ¿como hago para ponerle 16 pulsadores switch  para apagar y encender los relay manualmente?. 
[image]
[image]

tonhuisman, 2022-03-23:
(Translated using Google):

> Hi, I'm from Argentina, Buenos Aires. In my hobbies, I'm modernizing my smart home. I'm telling you that I'm an apprentice.
I have already successfully made the 4ch relay module work + mcu8266 + firmware esp easy + app inventor and with your help I placed 4 switch buttons to turn it off and on manually, using the rule.
now what I want is to use a 16-channel relay module and use the pca9685 module to control the 16 relays, and through the web browser the control of the relays works perfectly, the million dollar question is, how do I put 16 switch buttons on it to turn the relays on and off manually? 
1. Please state your questions in English, as that's the common language here
2. This is a question that better fits [the forum](https://www.letscontrolit.com/forum/index.php)

To answer your question: There is the `Keypad PCF8574 / MCP23017` plugin that can handle up to ~100 input buttons when connected in a matrix. Also, for that plugin there is an open PR #3927 to add support for the PCF8575 (16 I/O's) and a few more features. Also, there is an open PR #3964 that aims to support multiple input shift registers, waiting to be tested and merged.

TL;DR: Please try to use the Keypad plugin.

tonhuisman, 2022-03-26:
This support request has been [transferred to the forum](https://www.letscontrolit.com/forum/viewtopic.php?t=9065)

tonhuisman, 2023-09-09:
No reply, but solution provided, so can be closed.

closed 2023-09-09 by TD-er