letscontrolit/ESPEasy #3982: p2p - share task (at least bme280) from unit with unitnumber 0 fails

chromoxdor, 2022-03-30:
it took a while to figure out why sharing the sensor from one of my devices to another failed.
on the receiving device it looked like this:
[image]

by changing the unitnumber of the sending device it works:
[image]

tested all builds from 20116 and also different devices...

chromoxdor, 2022-03-30:
You are right.
[image]
But it is a bit confusing since the default value is 0.

closed 2022-03-30 by chromoxdor

tonhuisman, 2022-03-30:
> But it is a bit confusing since the default value is 0.

It is not a required setting, so what would you suggest for default? It 'only' has to be unique when using the P2P network 🤔

uzi18, 2022-03-30:
@tonhuisman ok but it should be more .... user friendly: 
- port number is wrong by default
- suggest/warn different unit number when 0 or 255 is used
- maybe something more?

know about docs but ... why it is so opposite user?

chromoxdor, 2022-03-30:
it shouldn’t be available as soon p2p is activated since it is stated in the documentation. something like a notification when enabling p2p would be helpful i guess.... 
....uzi18 was faster :)

uzi18, 2022-03-30:
@chromoxdor we are able to improve a bit experience of p2p network

chromoxdor, 2022-03-30:
The problem with unit 0 is, that it works like any other number except that task sharing doesn´t work (and there only sending...receiving works fine). So it not so obvious that the number causes the issue and therefore reading the p2p controller documentation is not the first thing that comes in mind.

TD-er, 2022-03-30:
Yep, it surely can use some extra warnings.
But like the warning for IDX=0 on Domoticz, even those errors in the logs lead to a number of questions.
So it is not very easy to make those warnings short and still very intuitive and clear.

uzi18, 2022-03-30:
we can catch situation when node send packets with id 0 and add another warning to change number to other

TD-er, 2022-03-31:
I guess such a warning next to the places where it can be set (both in UDP port setting and unit nr) is already an improvement.

uzi18, 2022-04-01:
@TD-er what do you think about #3989

tonhuisman, 2022-04-01:
Looks fine to me, I'll probably always see the new port-warning, as I'm using a different P2P port, but that's no problem 😃

uzi18, 2022-04-01:
@tonhuisman could you please test latest commit ( unit = 0 on root page) ? :)