letscontrolit/ESPEasy #3985: Unit Number always gets appended to hostname

chunter1, 2022-03-31:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

tonhuisman, 2022-03-31:
You have unchecked the `Append Unit Number to hostname` checkbox on the Config page?

chunter1, 2022-03-31:
Yes, exactly this is the issue.
It always gets added - and if you check the box it get added even a second time like.... "UNITNAME_12_12"

tonhuisman, 2022-03-31:
Where do you see it getting added? (I have an idea, but like to see your exact finding... 😃)

TD-er, 2022-03-31:
Just as an example, with a node I just updated to the latest source
[image]

Where is it added twice?

TD-er, 2022-03-31:
Hmm I do see it is added twice on the MQTT Client ID:

[image]

chunter1, 2022-03-31:
I see it in FHEM under the MQTT devices.

TD-er, 2022-03-31:
As the default client ID, like in my screenshot?

chunter1, 2022-03-31:
Yes

TD-er, 2022-03-31:
Is that the only place?

chunter1, 2022-03-31:
I noticed it only there since i an only usinf FHEM with MQTT.

tonhuisman, 2023-09-09:
This is a fix I have tested some time ago, and confirmed working, so this can most likely be closed.

closed 2023-09-09 by TD-er