letscontrolit/ESPEasy #3988: Enable log in ESPeasy

jvdVlugt, 2022-04-01:
I want logging to be enabled in platformio for ESPEASY but don't know how to get this done. 
Now when I add /log in the browser it tell me "Not included in build"
Already tried to find answeres in the internet without success.
Hopefully someone can help we on this.

TD-er, 2022-04-01:
You can pick the "upload and monitor" option in PlatformIO.

Just a pick from my PIO option selection in VScode:
[image]

When using Atom, these options are also available, only slightly less easy to select.
For each PIO env. you can select an option with "monitor" or "upload and monitor".
The log via the serial port will then be shown in the console.
Even better, when using VS code at least, you will yet a proper crash dump if it crashes, so you get some hint on what may have happened in what part of the code.

jvdVlugt, 2022-04-01:
Not sure this is what a need.
I mean the logging that is accessable trough the "Tools / system" menu in espeasy from a normal built from a bin file but now I want that enabled on my own built

tonhuisman, 2022-04-01:
If you add a line with 
`#define WEBSERVER_LOG`
to your `Custom.h` file you should be good to go

jvdVlugt, 2022-04-01:
Thanks. That did the trick

tonhuisman, 2022-05-07:
This seems resolved, please close the issue.

closed 2022-05-07 by TD-er