letscontrolit/ESPEasy #3991: CSE7766 does not work on latest esp8266 core

TD-er, 2022-04-02:
The code for this plugin has hardly changed the last year, but now on the latest core build, it no longer seems to work.

This plugin does some strange things with the Serial0 port, so that's probably what's wrong here.

TD-er, 2022-04-25:
See also [this forum thread](https://www.letscontrolit.com/forum/viewtopic.php?p=58219#p58219).
These changes are also required to make this plugin work on ESP32.

tonhuisman, 2022-11-09:
This is most likely resolved by #4328

tonhuisman, 2023-02-11:
Removed the `strange things with the Serial0 port` part from the plugin in #4501 that might solve this issue. Tested that PR with `hard_Sonoff_POW ESP8266` build, and works as intended.

PR #4501 (2023-02-13): [P077] Feature: CSE7766 add serial GPIO pin configuration
Resolves #4473
Resolves #4502 
Resolves #3991 
and some Forum requests (f.e.: [1](https://www.letscontrolit.com/forum/viewtopic.php?t=8417), [2](https://www.letscontrolit.com/forum/viewtopic.php?t=9525))

Features:
- Add serial GPIO pin configuration

TODO:
- [x] Test on Sonoff POW R2 (ESP8266), supports both HWSerial0 and SoftwareSerial.
- [x] Add commands to set the calibration values.
- [x] Add documentation.
- [x] Testing Sonoff Origin (POWR3xx) ([Confirmed](https://github.com/letscontrolit/ESPEasy/issues/4473#issuecomment-142676487 [... cut]
closed 2023-02-13 by TD-er