letscontrolit/ESPEasy #3992: [P096] Add support for Waveshare 1.54"200x200 display

tonhuisman, 2022-04-02:
As a previous pull request was taking too much time, combined with development of this display, this is pulled separate to be done when possible (a work in progress, already)

kretzp, 2022-04-02:
I have a Waveshare 1.54" 200x200 display. If you want me to help to develop it, I will do it. But then I suppose it will go to P116 like Waveshare 2.7" right?

tonhuisman, 2022-04-02:
I already have the code partially complete, just waiting for a unit to arrive (and that will take a few weeks) I'll create a PR in the coming days, so you can start testing 👍

kretzp, 2022-04-03:
Yea sorry, P096 is of course rigth. Ok please let me know if you have the PR, then I will start testing.