letscontrolit/ESPEasy #4000: PMS 5003 NO Values

FaresEdres, 2022-04-09:
My Circuit:

[image]

Code:

#include <SoftwareSerial.h>
SoftwareSerial pmsSerial(14, 15);
 
void setup() {
  // our debugging output
  Serial.begin(115200);
 
  // sensor baud rate is 9600
  pmsSerial.begin(9600);
}
 
struct pms5003data {
  uint16_t framelen;
  uint16_t pm10_standard, pm25_standard, pm100_standard;
  uint16_t pm10_env, pm25_env, pm100_env;
  uint16_t particles_03um, particles_05um, particles_10um, particles_25um, particles_50um, particles_100um;
  uint16_t unused;
  uint16_t checksum;
};
 
struct pms5003data data;
    
void loop() {
  if (readPMSdata(&pmsSerial)) {
    // reading data was successful!
    Serial.println();
    Serial.println("---------------------------------------");
    Serial.println("Concentration Units (standard)");
    Serial.print("PM 1.0: "); Serial.print(data.pm10_standard);
    Serial.print("\t\tPM 2.5: "); Serial.print(data.pm25_standard);
    Serial.print("\t\tPM 10: "); Serial.println(data.pm100_standard);
    Serial.println("---------------------------------------");
    Serial.println("Concentration Units (environmental)");
    Serial.print("PM 1.0: "); Serial.print("data.pm10_env");
    Serial.print("\t\tPM 2.5: "); Serial.print(data.pm25_env);
    Serial.print("\t\tPM 10: "); Serial.println(data.pm100_env);
    Serial.println("---------------------------------------");
    Serial.print("Particles > 0.3um / 0.1L air:"); Serial.println(data.particles_03um);
    Serial.print("Particles > 0.5um / 0.1L air:"); Serial.println(data.particles_05um);
    Serial.print("Particles > 1.0um / 0.1L air:"); Serial.println(data.particles_10um);
    Serial.print("Particles > 2.5um / 0.1L air:"); Serial.println(data.particles_25um);
    Serial.print("Particles > 5.0um / 0.1L air:"); Serial.println(data.particles_50um);
    Serial.print("Particles > 10.0 um / 0.1L air:"); Serial.println(data.particles_100um);
    Serial.println("---------------------------------------");
  }
}
 
boolean readPMSdata(Stream *s) {
  if (! s->available()) {
    return false;
  }
  
  // Read a byte at a time until we get to the special '0x42' start-byte
  if (s->peek() != 0x42) {
    s->read();
    return false;
  }
 
  // Now read all 32 bytes
  if (s->available() < 32) {
    return false;
  }
    
  uint8_t buffer[32];    
  uint16_t sum = 0;
  s->readBytes(buffer, 32);
 
  // get checksum ready
  for (uint8_t i=0; i<30; i++) {
    sum += buffer[i];
  }
 
  /* debugging
  for (uint8_t i=2; i<32; i++) {
    Serial.print("0x"); Serial.print(buffer[i], HEX); Serial.print(", ");
  }
  Serial.println();
  */
  
  // The data comes in endian'd, this solves it so it works on all platforms
  uint16_t buffer_u16[15];
  for (uint8_t i=0; i<15; i++) {
    buffer_u16[i] = buffer[2 + i*2 + 1];
    buffer_u16[i] += (buffer[2 + i*2] << 8);
  }

Result:
[image]

TD-er, 2022-04-09:
Your `readPMSdata` function does not seem to ever return a `true` value.
Apart from that, I'm not sure what to do with this issue as it isn't about ESPEasy.
In ESPeasy there is already a plugin for these sensors which is working fine.

FaresEdres, 2022-04-09:
> Your `readPMSdata` function does not seem to ever return a `true` value. Apart from that, I'm not sure what to do with this issue as it isn't about ESPEasy. In ESPeasy there is already a plugin for these sensors which is working fine.

so it's a code problem not circuit or the sensor?
and what do u think i have to change in the code?
thanks in advance

TD-er, 2022-04-09:
I don't know if that's the only issue.
But like I said, I don't know what to do with this issue as it is not an ESPEasy issue, not running on a platform supported by ESPEasy and the mentioned sensor is supported in ESPEasy.
So please tell me what I am supposed to do with this issue.

FaresEdres, 2022-04-09:
> I don't know if that's the only issue. But like I said, I don't know what to do with this issue as it is not an ESPEasy issue, not running on a platform supported by ESPEasy and the mentioned sensor is supported in ESPEasy. So please tell me what I am supposed to do with this issue.

what is ESPEasy?

TD-er, 2022-04-09:
That's the firmware being developed in this GitHub project to which you posted your issue.

closed 2022-04-09 by TD-er