letscontrolit/ESPEasy #4008: Shelly plus 1pm

EDSTOBI, 2022-04-17:
Hello and happy Easter,
has anyone already got a Shelly plus 1PM running with ESPasy and gets the correct current values?
Or should I buy the 1PM model?
THX
EDSTOBI

tonhuisman, 2022-04-17:
I also bought a Shelly 1PM and installed ESPEasy on it. With the same result, not providing Current and Voltage readings. After some searching I found an issue at the Tasmota project, with a description how to hardware-patch the unit to (try to) enable these readings. _I_ didn't succeed, as you can [read in the forum](https://www.letscontrolit.com/forum/viewtopic.php?p=45771), most likely caused by my lack of micro-soldering skills 😄, but others seem to have that working.
It does give a reliable Power reading though (once 'calibrated'), and is a _really_ stable unit (running for 268 days, since the power had to be cut due to a utility-meter replacement)

The short answer: Shelly 1PM doesn't have the connections wired to read the current.
The long answer: _Maybe_ you can modify the hardware to read current and voltage.

The main difference between the 'regular' and the 'Plus' model is that the Plus has an ESP32 with 4MB Flash, where the regular is ESP8266 based with 2MB Flash. Here's a [nice comparison](https://peyanski.com/shelly-plus-1-and-shelly-plus-1pm-vs-old-shellies/)

tonhuisman, 2022-04-17:
I forgot to mention before, but this would have been an ideal subject for the forum...

tonhuisman, 2023-09-09:
This seems to be resolved, so can be closed.

closed 2023-09-09 by TD-er