letscontrolit/ESPEasy #4013: WIFI problem esp32 wroom

neolead, 2022-04-22:
Heeey, what can i do..??
used esp32..
all owrks ok..
but connect to wifi 1 of 20 times?!!!

with reserved second AP ...same..
if connected.. no problem

TD-er, 2022-04-22:
I think I need a bit more info...
Like what build are you using, etc.

Just to make sure it isn't already fixed in the latest code (not yet built a new official build with it), you may want to check the GitHub Actions test build from [here](https://github.com/letscontrolit/ESPEasy/actions/runs/2207014011)

Just use the "...factory.bin" builds when flashing via serial and the one without "factory" in the name for upgrading via the web interface.

laimerIT, 2022-04-24:
Hi,
i have the same problem with different ESP32 and tested the last 4 normal and test A versions. One ESP32 is sending data every second, if WiFi ist on load it works.
ESPEasy with low work has the problem, the GUI ist very slow.
Look at the Ping Example below, if i upload a Firmware ther is no timeout and every Ping ins unser 200ms.
Greetings from Bavaria
```
Antwort von 192.168.10.74: Bytes=32 Zeit=2086ms TTL=255
Antwort von 192.168.10.74: Bytes=32 Zeit=11ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.10.74: Bytes=32 Zeit=3090ms TTL=255
Antwort von 192.168.10.74: Bytes=32 Zeit=4ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.10.74: Bytes=32 Zeit=15ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.10.74: Bytes=32 Zeit=3438ms TTL=255
Antwort von 192.168.10.74: Bytes=32 Zeit=2ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.10.74: Bytes=32 Zeit=23ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.10.74: Bytes=32 Zeit=2066ms TTL=255
Antwort von 192.168.10.74: Bytes=32 Zeit=6ms TTL=255
```

neolead, 2022-04-24:
Tomorrow will logs it here.. just not connect... sometimes if 10 reboot it. Just hang it up by teeth. And work month..

TD-er, 2022-04-24:
Just a few things to consider:

- Make sure "Eco" mode is off
- Check to send with max TX power
- Do you have several access points with the same name -> run latest build from GitHub Actions test builds as this might be fixed.
- Please give some indication on the RSSI value reported by ESPEasy

neolead, 2022-04-26:
than...
Build: ESP_Easy_mega_20211224_display_ESP32_4M316k Dec 24 2021
connected with thermo sensor, encoder, and lcd...

neolead, 2022-04-26:
1.eco mode off..
2.send with max TX power
3. tried and several with same... and exclusive.. and added separate ap for this esp..
4. max wifi tx 17.00
current wifi tx 15
Send With Max TX Power : true
Restart WiFi Lost Conn:false

neolead, 2022-04-26:
cannot change build... because this lcd are working ok with this build

neolead, 2022-04-26:
[image]

neolead, 2022-04-26:
[image]

neolead, 2022-04-26:
[image]

neolead, 2022-04-26:
[image]

neolead, 2022-04-26:
[image]

TD-er, 2022-04-26:
Maybe your setup also experiences similar issues to what's described here: https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-1109132101

Can you try one of the latest test builds for the upcoming next official build? See: https://github.com/letscontrolit/ESPEasy/actions/runs/2216433073

N.B. you need to be logged in to be able to download the ZIP file.

Jason2866, 2022-04-27:
@TD-er  Have you tried with latest Arduino Core 2.0.x ?
There are many many changes / fixes done in IDF4.4. regarding wifi?
Imho it is time to say goodbye to the outdated not anymore supported Arduino core 1.0.x Searching bigs there is wasted time, since even when you found them, the wont be fixed.

TD-er, 2022-04-27:
@Jason2866 Yep, I have a few test nodes here running IDF 4.4
The WiFi has indeed been improved, but some things regarding Ethernet are now working differently.
Still have to check if the issue is now fixed where DHCP may only set the IP/subnetmask/gateway and not the DNS.
This is a problem with the older core.
But the newer core can no longer see whether a link is up.

sobolkz, 2022-05-15:
Same issue with Wemos D32 Pro and WiFi connection after deep sleep waking up. Expected with mega-20220328 and mega-20220427. Board can be booted with Reset button at board.

TD-er, 2022-06-24:
Not really sure if this is still a problem (and if it is, what might be the cause)
Can you try the latest official build, or one of the latest GitHub Actions builds, like this one: https://github.com/letscontrolit/ESPEasy/actions/runs/2550549920

sobolkz, 2022-06-24:
> Not really sure if this is still a problem (and if it is, what might be the cause) Can you try the latest official build, or one of the latest GitHub Actions builds, like this one: https://github.com/letscontrolit/ESPEasy/actions/runs/2550549920
@TD-er, it isn't actual for me. WiFi working perfect with May's and newest builds.

TD-er, 2022-06-24:
Thanks for the feedback!

closed 2022-06-24 by TD-er