letscontrolit/ESPEasy #4020: [ESP32] Detect single core 160 MHz ESP32-U4WD (Shelly Plus i4)

TD-er, 2022-04-25:
The Shelly Plus i4 seems to be using the ESP32-U4WD
This one has 4MB embedded flash.
But on chip silicon from before 2021/12/02, this ESP32 only had 1 core running at max. 160 MHz.

Jason2866, 2022-04-27:
If i read the product change sheet from espressif correctly it does support 240 Mhz.
Anyways we use for Tasmota as standard 80 Mhz clock speed. All supported Tasmota Shelly devices with the single core ESP32 do run stable with our single core Arduino32 framework.
You can NOT use a standard usual Arduino framework for any single core ESP32.

TD-er, 2022-04-27:
@Jason2866 Thanks for the heads up.
That will probably save me a lot of trial and error.

About the 240 MHz... the strange thing is that the earlier revisions apparently did support only 160 MHz, but after the update it supports 240 MHz. (and dual core)

tonhuisman, 2023-09-09:
This seems to be solved, so can be closed.

closed 2023-09-09 by TD-er