letscontrolit/ESPEasy #4021: Cannot save .dat files from one ESP8266 node

ghtester, 2022-04-25:
Maybe it's not a fresh issue as I am trying to save the config files for the first time from the affected ESP Easy node.

The same issue happens when I try to:
- save the config file through the Tools - Settings -Save button
- save any .dat file from File browser

Tested from Firefox and Edge web browsers with the same result - it ends with timeout and the .dat file is not saved. 
The rule files (.txt) can be saved from File browser without issue.

The ESP node was upgraded several times from earlier ESP Easy builds and everything (except the .dat files saving) looks working fine.
The firmware is Custom build, compiled 2022-04-22. Other ESP Easy nodes with the same latest build (but with different plugins/controllers configuration) does not have that issue.

Any hint I could try before the complete config reset to Factory defaults please?

TD-er, 2022-04-25:
First thing that comes to mind is file system fragmentation.
But that should only happen on tight file systems or file systems which are quite full.

What you can do, is remove the rules files (back them up somewhere) and reboot the ESP.
At reboot, before mounting, ESPEasy will call the file system garbage collector, which will then free up blocks that have either free or no longer allocated pages in it.
Then make some change in the settings to force a write and reboot it again to repeat it.
After that you might be able to upload the rules files again.

What kind of file system size do you have and how much is it filled?

ghtester, 2022-04-25:
Thanks a lot for the advice, I'll give it a try.
I don't have an access to that ESP node right now so the filesystem details I could share later.
But it's 4M node and I doubt there's a lot of space occupied.
The rule files are also small, AFAIK 3 of 4 are currently used.

TD-er, 2022-04-25:
Have you exceeded the nr of flash writes of the day?
This counter can be reset:  `ResetFlashWriteCounter`

ghtester, 2022-04-25:
Not at all, there were no recent config changes before the first config save attempt. Then I tried to make some config changes (to disable Rules Cache feature) without any positive impact.

ghtester, 2022-04-25:
`Thanks again for hint, I can confirm it helped. 
I have saved largest rules3.txt file (882 bytes), deleted it, rebooted, made a slight change in settings & rebooted again. After that I was able to save the configuration.  Then I uploaded the rules3.txt back. Rebooted and tried to save all files from the File browser. Despite I have experienced Exceptions & unexpected reboots, after several repeats I could finally save all .dat files.

closed 2022-04-25 by ghtester